tomquist/AstraMeter #1: TypeError: 'type' object is not subscriptable

gine78, 2024-06-21:
Hello, thank you for your work here. I just try your code but i get a error. I have try it with mqtt an Homeassitant, but i always get this Error:

```
~/b2500-meter$ pipenv run python smartmeter.py
Traceback (most recent call last):
  File "smartmeter.py", line 22, in <module>
    class Powermeter:
  File "smartmeter.py", line 27, in Powermeter
    def get_powermeter_watts(self) -> tuple[int, ...]:
TypeError: 'type' object is not subscriptable
```

Thank you

tomquist, 2024-06-21:
What does your config look like and what is the topic payload?

gine78, 2024-06-21:
hey, my config file:
`[GENERAL]
# Sum the power values of all phases and report them as a single value on phase 1 (True/False)
DISABLE_SUM_PHASES = False
# Allow negative power values (True/False)
ALLOW_NEGATIVE_VALUES = False
SKIP_POWERMETER_TEST = False

[MQTT]
BROKER = 192.168.188.15
PORT = 1883
TOPIC = tele/tasmota_B86xxx/SENSOR
JSON_PATH = x.SML.curr_w
USERNAME = xxx
PASSWORD = xxx
`
The rest is comment out

gine78, 2024-06-21:
Ah, i forgot:

`tele/tasmota_B86XXX/SENSOR {"Time":"2024-06-21T22:09:59","SML":{"server_id":"0a01484c5902001d7cXX","export_total_kwh":0.0000,"total_kwh":3962.1601,"curr_w":381,"volt_p1":0.0,"volt_p2":0.0,"volt_p3":0.0,"amp_p1":0.0,"amp_p2":0.0,"amp_p3":0.0,"phase_angle_l2_l1":0.0,"phase_angle_l3_l1":0.0,"phase_angle_p1":0.0,"phase_angle_p2":0.0,"phase_angle_p3":0.0,"freq":0}}`

tomquist, 2024-06-21:
The JSON_PATH is not valid. Can you try `$.SML.curr_w` instead?

gine78, 2024-06-21:
I just have try allready, i got the same error. 

I have also try it with 

```
[HOMEASSISTAN]
IP = 192.168.188.148
PORT = 8123
HTTPS = False
ACCESSTOKEN = eyJhbGciOiJIUzI1NiIsIXXXXXXXXXXXXXXXXXXXXusw
CURRENT_POWER_ENTITY = sensor.stromzahler
POWER_CALCULATE = True
POWER_INPUT_ALIAS = sensor.power_input
POWER_OUTPUT_ALIAS = sensor.power_output
```

it ends with the same error.

tomquist, 2024-06-22:
Which Python version are you using? Do you use the latest version of the script?

gine78, 2024-06-22:
`~/b2500-meter$ python3 --version
Python 3.8.10`

i run

```
pipenv run python3.8 smartmeter.py
Traceback (most recent call last):
  File "smartmeter.py", line 22, in <module>
    class Powermeter:
  File "smartmeter.py", line 27, in Powermeter
    def get_powermeter_watts(self) -> tuple[int, ...]:
TypeError: 'type' object is not subscriptable
```

i dont know what false here i just follow you instructions in the readme file. I use a ubuntu enviroment.

gine78, 2024-06-22:
yes, i try the last version. (git fetch)

gine78, 2024-06-22:
next step:

i have set the std. python to 3.8.10, now i run in this error

```
~/b2500-meter$ pipenv run python smartmeter.py
Warning: Your Pipfile requires python_version 3.7, but you are using 3.8.10 (/home/xxx/.local/share/v/b/bin/python).
  $ pipenv check will surely fail.
Testing powermeter configuration...
Error: 'state'

```

tomquist, 2024-06-22:
Can update to the latest version and try again?

gine78, 2024-06-22:
I remove an download the new one, same problem like before.

My python version is 3.8.10 i follow the instruction, now i delete the whole directory and make a new one (i save the config file)

```
git clone git clone https://github.com/tomquist/b2500-meter.git
cd b2500-meter
pipenv install
pipenv run python smartmeter.py
xxx:~/b2500-meter$ pipenv run python smartmeter.py
Testing powermeter configuration...
Error: 'state'
```
can you explain me, what entities are the following are?

```
POWER_INPUT_ALIAS = sensor.power_input
POWER_OUTPUT_ALIAS = sensor.power_output
```
what is your script waiting for it here?

gine78, 2024-06-22:
Update: 

```
MQTT is workinng well, i think it is just an HOMEASSISTANT error? 
~/b2500-meter$ pipenv run python smartmeter.py
smartmeter.py:356: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  self.client = mqtt.Client()
Testing powermeter configuration...
Connected with result code 0
Successfully fetched powermeter value: 3993W
TCP server is listening...
Received 'hame' from ('192.168.188.111', 10000), sent 'ack'
TCP connection established with ('192.168.188.111', 10000)
Received 'hello'
Sent message: HM:2843|0|0
Sent message: HM:2843|0|0

```

tomquist, 2024-06-22:
Are you sure it's exactly the same error as before? If so, you probably didn't update to the latest version.

`POWER_INPUT_ALIAS` should be the meter measurement for incoming power from the grid and `POWER_OUTPUT_ALIAS` the outgoing power value.

gine78, 2024-06-22:
I now use MQTT it works well and my BP2500 receive the measurement very quick - but the reacion is like a "Bahnschranke" :-)

tomquist, 2024-06-22:
Yeah, unfortunately it’s quite unusable as it is. Let’s hope it’s getting better with a future firmware.

gine78, 2024-06-22:
Jeah, Thank you for your Work. 

_Und entschuldige mein miserables englisch... hab aber auch kein bock das immer bei deeple reinzuhauen._

closed 2024-06-25 by tomquist