tomquist/AstraMeter #3: Interpretation of negative Values in b2500

olympic1337, 2024-06-24:
Hi @tomquist,
many thanks for your great work here!
I tested negative values with my v3 b2500 using firmware v210. This leads to an integer overflow in the b2500 device causing strange behavior in the power output. A reboot of b2500 seems to solve the issue again. Shall I place a pull request to slightly adapt the docu accordingly?

Best regards, looking forward to seeing more of that great progress here!

tomquist, 2024-06-24:
Yeah I'm aware of this issue. That's why negative values rounded to 0 unless you have `ALLOW_NEGATIVE_VALUES` enabled. I already have this in the docs:
```
# It looks like the B2500 storage currently does not support negative power values so they get clamped to 0 by default. To disable this behavior, you can set the following configuration to `True`
```
However,  this could probably be pointed out a bit more explicitly.

lutz-fechner, 2025-05-25:
Are you sure this is working? I have my 3 line power values as scv in "Power Input Alias" and "Allow Negative Values" turned off, but I see this in the log:

`
Sending response: {"id":0,"src":"shellypro3em-ec4609c439c1","dst":"unknown","result":{"a_act_power":113.6,"b_act_power":-198.0,"c_act_power":53.5,"total_act_power":-30.9}}
Received UDP message: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
From: 192.168.178.65:22222
Parsed request: {
  "id": 0,
  "method": "EM.GetStatus",
  "params": {
    "id": 0
  }
`

tomquist, 2025-05-25:
The "Allow Negative Values" is only relevant for the CT001 emulator. For Shelly, negative values are supported out of the box.

tomquist, 2025-05-25:
No, there's nothing to worry about. Hame didn't correctly implement JSON-RPC because they don't send a `src` parameter, which is the reason for the `unknown`, but the storage ignores it anyway.