tomquist/AstraMeter #4: Issue with Homeassistant poller

rschoell, 2024-07-26:
Hi Tom, great work you've done here. 
I've managed to get it running within my docker environment, but am running into some issues.
My configuration looks as follows:
[image]
the sensor "inout" is a summation as you would do with the input output subtraction. 
So in case of power overflow this sensor would go into a negative. 

I'm able to get the first value from the test as well as a connection from the B2500 to the virtual meter. 
But no actual data is getting queried by the meter in the polling interval.
[image]

I know this is not yet working on Marstek's side, but I would love to ongoing use your implementation.

noone2k, 2024-07-26:
i see different networks ... are the devices in different subnets ?
some router / sw-implementations allow udp broadcasts to passthrough, but direct messages are blocked.
looks like either the udo reply is blocked or the tcp traffic between ct001 and b2500 ... 
or the NAT  has a problem ( may be a docker problem, but i dont use any containers. have to be answered by someone with more knowledge in this subject )

btw. another possibility: the used firmware on the b2500 ist to old ...

tomquist, 2024-07-26:
Yeah, to run this in Docker you'll likely have to use the host network for the docker container. I'm not sure it's easily possible otherwise as the storage is trying to reach out to the host that sent the UDP package. Also: did you open up TCP port 12345 on the docker container?

rschoell, 2024-07-26:
Thanks for your prompt answer. I've opened both TCP and UDP.
NAT was also my first point on where I would assume it could hang currently. I'll prepare a direct linux container on the host network and see, how things go.

And to the point of @noone2k, I believe 2.14 is the current latest for v2 batteries. I received last Friday an update to the 2.15 beta, but scrapped it and downgraded again, as I was no more able to control output.
It was only on 800W and wasted lots of energy on the sunny days. 😉

rschoell, 2024-07-26:
@tomquist ok, so running it within the same network seems better, although this also kind of applied with docker. 
the docker container has a nated IP, but the port is opened on the same network as home assistant. However yet not able to get it working that way. 
But like we already assumed, even though these values from the same subnet machine are being submitted well, and the Marstek /PowerZero apps show the correct value of the current meter, nothing moves in terms of output and if so, it does it on wrong values. 
Also the battery at that time seems incapable of running the PV ports.

olympic1337, 2024-07-27:
@rschoell The smart-meter-related software in the battery pack seems to be a mess. I've experienced the same behavior, also what you mentioned for the PV ports. Answer of hame was that it only works with original smart meters (which should for my understanding behave the same like tomquisits software does.
I finally switched over to using opendtu and control the power output on the inverter.

rschoell, 2024-07-29:
@olympic1337 yes, I read similiar reports on the internet as well. Still a to be corrected product. I'll be switching to the same method and will control this ongoing through the inverter.