tomquist/AstraMeter #7: Proposal: sending absolute values seems to work with V216

mikj77, 2024-09-23:
Hello Tom,
thanks a lot for this great project.
The b2500 seems to work with V216 if you provide absolute values rather than cutting off negative values.
By changing b2500.py as follows, the battery seems to compensate the power properly:
##                    value1 = max(value1, 0)
##                    value2 = max(value2, 0)
##                    value3 = max(value3, 0)
                    value1 = abs(value1)
                    value2 = abs(value2)
                    value3 = abs(value3)
Via trial and error b2500 finds out whether the value is in fact positive or negative and then adapts slowly but correctly.
Best regards
Michael

tomquist, 2024-09-23:
The project is doing that already. The `abs()` function stands for „absolute“ and is doing the same as `max(value, 0)`.

closed 2024-09-24 by mikj77

LagaV, 2024-10-07:
> The project is doing that already. The `abs()` function stands for „absolute“ and is doing the same as `max(value, 0)`.

Are you sure on this? AFAIK abs(-5) results in 5 and max(-5,0) results in 0. The above observation from @mikj77 might be relevant.

tomquist, 2024-10-11:
Oh 🤦 I think I initially misunderstood the suggestion. I thought the suggestion was to switch to `max`. I already updated the Node-RED flow to use `abs` in June but it looks like I forgot to do the same change to the Python script and just assumed I did.

Anyway, I just updated the Python script to also use `abs`, which is definitely more correct than using `max`.

commit f91fb2e1 (2024-10-11): Update Python app to send absolute value

Fixes #7