tomquist/AstraMeter #8: HA-Addon scheint nicht zu funktionieren

ugath, 2024-10-12:
Habe alles wie in der Anleitung gemacht: Addon installiert, konfiguriert und gestartet.
Scheint auch zu laufen aber die Marstek-App findet kein CT001.
[image]
[image]
[image]
Home Assistant ist aktuell.

tomquist, 2024-10-12:
Ja, habe auch gesehen, dass die neuste Marstek diese Meldung anzeigt. Ich weiß nicht genau was die App da macht, jedenfalls sieht man in den Logs des Addons dass sich der Speicher verbindet (zumindest bei mir). Was zeigen die Logs bei dir an?

Versuch auch mal die App zu schließen und neu zu öffnen, dann nur auf Selbstanpassung stellen (ohne den blauen Bitton zu drücken).

ugath, 2024-10-12:
Im Protokoll ist eine Fehlermeldung:

> [HOMEASSISTANT]
> IP=supervisor
> PORT=80
> API_PATH_PREFIX=/core
> ACCESSTOKEN=c21fa354d6f3a22fb00fefb9ae9d036f4acca46face6abc1131175464d76123c16c8b6fa377454d2a8a24d3c1ea8021f3460cb8c8660afb9
> POWER_CALCULATE=False
> CURRENT_POWER_ENTITY=sensor.electricity_XXXXXXXXXXXXX_gesamtleistung
> Testing powermeter configuration...
> **Error: invalid literal for int() with base 10: '29.63'**

tomquist, 2024-10-12:
Ah, liegt wahrscheinlich am Punkt in dem Sensorwert. Muss ne Ganzzahl sein. Kann ich später beheben.

tomquist, 2024-10-12:
Ok, eben eine neue Version gepushed. Kannst du es nochmal probieren? Am Besten erst die alte Version deinstallieren, dann nochmal rüber kopieren und dann die neue Version installieren.

ugath, 2024-10-13:
Super, jetzt funktioniert es... Danke!
Jetzt müsste Hame nur noch eine vernünftige "Selbstanpassung" hin bekommen. So wie es jetzt ist, ist es ja so gut wie unbrauchbar. ;)
Hier hat sich die Entladung und damit auch die Einspeisung immer weiter aufgeschaukelt. Hat doch nichts mit der Emulation zu tun, oder (hatte dann wieder auf Zeitsteuerung umgeschaltet, bevor ich noch mehr verschenke)? 
[image]

tomquist, 2024-10-13:
Ich würde nicht ausschließen, dass es an der Emulation liegt. Habe selbst keinen CT001 und kann daher nicht vergleichen. So wie ich allerdings gehört habe, ist es mit dem Original auch nicht viel anders. Aktuell ist noch ein bisschen unklar, wie der CT001 mit negativem Verbrauch umgeht. Fakt ist, dass der Speicher es gar nicht mag, wenn man ihm negative Werte schickt. Dann kommt es zu einem Integer Overflow und die App zeigt einen Verbrauch von 65500W an. Die Theorie ist daher, dass der CT001 den Absolutwert schickt, was ich im Emulator so abgebildet habe.

closed 2025-01-29 by tomquist