tomquist/AstraMeter #13: Support for Marstek Venus

matrover, 2025-01-21:
Hi Tom, 

First of all, great work! 
Long shot here, but i own a (new) Marstek Venus 5,12kwh. it came with a CT001 but thats not working great so i tried your HA Add-on. The batt seems to find the add-on (according to the logs) but it doesnt seem to complete the handshake. 

Do you have any pointers for me to test? 

Log:
```
Received 'hame' from ('192.168.1.110', 22222), sent 'ack'
Received 'hame' from ('192.168.1.110', 22222), sent 'ack'
Received 'hame' from ('192.168.1.110', 22222), sent 'ack'
Received 'hame' from ('192.168.1.110', 22222), sent 'ack'
Received 'hame' from ('192.168.1.110', 22222), sent 'ack'
```

(restarted the batt, app and HA + Addon) 

Core 2025.1.3
Supervisor 2024.12.3
Operating System 14.1
Current version: main (downloaded 21-01-25)

matrover, 2025-01-22:
Maybe interesting if i select CT002 & CT003 it gives these errors: 

```
Received unknown UDP message: 49|HMG-1|2419720d3e32|HME-4|000000000000|0|02
Received unknown UDP message: 49|HMG-1|2419720d3e32|HME-4|000000000000|0|02
Received unknown UDP message: 49|HMG-1|2419720d3e32|HME-4|000000000000|0|02
```

matrover, 2025-01-23:
This is great news Tom! thanks.

I'll try to see if i can run it directly to rule out the HA or docker issues. 
We think it doesnt work because the Venus and CT001 are communicating via BLE. but that a wild guess. 
The CT002 & 003 work via (W)Lan on the Venus

tomquist, 2025-01-25:
Once this change lands, this project will also be able to emulate a Shelly device: https://github.com/tomquist/b2500-meter/pull/16

Would you be able to test if it works on the Marstek Venus?

matrover, 2025-01-25:
Yes i have this option to test it in marstek, let me know when i can test!

tomquist, 2025-01-25:
It landed. You can configure the Shelly Pro 3EM emulator through the `DEVICE_TYPE` setting (in the `[GENERAL]` section) in your `config.ini`, e.g.
```
[GENERAL]
DEVICE_TYPE=shellypro3em
```

matrover, 2025-01-25:
Like this in HA right? 

[image]

It doenst show anything in the logs. 
the app says: Config failed, try again

tomquist, 2025-01-25:
I think you're still running the old image. Can you try again, updating to version 1.0.0 I just pushed? It should offer you an update after refreshing the addon store.

tomquist, 2025-01-25:
Oh, actually it might take a couple more minutes until the version is ready

matrover, 2025-01-25:
1.0.1 doesnt start on HA: 

```
Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Traceback (most recent call last):
  File "/app/main.py", line 170, in <module>
    main()
  File "/app/main.py", line 150, in main
    powermeter = create_powermeter(cfg)
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/config/config_loader.py", line 62, in create_powermeter
    raise Exception("Error: no powermeter defined!")
Exception: Error: no powermeter defined!
```

kdhoe85, 2025-01-25:
Thanks @tomquist the Shelly emulator works great!

tomquist, 2025-01-25:
@kdhoe85 Looks like it’s working for you. Are you using it through Home Assistant or are you running the Docker image/Python script directly?

tomquist, 2025-01-25:
@matrover I found the issue. Can you please update to 1.0.2 and try again?

kdhoe85, 2025-01-25:
> [@kdhoe85](https://github.com/kdhoe85) Looks like it’s working for you. Are you using it through Home Assistant or are you running the Docker image/Python script directly?

I’m running it via Docker

matrover, 2025-01-26:
Yes works with 1.0.2 on HA! many thanks :) 

The Venus (auto mode) is not really working well on the readings but i dont think it is due to this addon. 
Its charging and discharging all the time, while there is enough solar (growatt) to charge 

[image]

[image]

[image]

tomquist, 2025-01-26:
It’s possible the power usage sensor in HA isn’t updating fast enough. Which smart meter are you using? Try increasing the polling interval for your original smart meter.

matrover, 2025-01-26:
Im using a Youless LS120 which updates ever 10secs. 
Now reconfigured it to the Homewizard P1 dongle which updates every 5secs, lets see. 

Any recommendations in terms of update speed?

tomquist, 2025-01-26:
I don’t have specific recommendations. The faster the better. I think the Marstek requests new measurements around once a second so ideally the power sensor would update once a second as well.

matrover, 2025-01-26:
Yes with 5secs it works! 
Many thanks (again :))

[image]

tomquist, 2025-01-26:
Awesome! Glad this worked. I’ll close the issue for now.

closed 2025-01-26 by tomquist

matrover, 2025-01-28:
updated to 1.0.2 @Kolbi