tomquist/AstraMeter #24: Power Offset und Power der Phasen

Ultraschall81, 2025-01-30:
Hallo Tom,

Ich habe seit heute den Marstek Venus und habe dein Tool in HA installiert. Den Shelly 3em habe ich schon lange im Zählerkasten verbaut.
Funktioniert auf Anhieb super, vielen Dank schonmal für deine Arbeit.

Zwei Improvements hätte ich.
1. (prio1) Mein Shelly 3em zeigt in summe ca 20 Watt weniger an als mein Hauszähler. Ich fände einen Power offset Regler wäre eine super Sache. So kann man die Ungenauigkeiten zwischen Hauptzähler und Zusatzzähler (Bsp Shelly) eliminieren und gleichzeitig einen Offset einstellen. (Bsp: Ich will immer 10W Bezug oder 10 W einspeisen)

2. (prio2) Derzeit nimmt er die einzelnen Werte der Phasen vom Shelly 3em und summiert sie auf Phase 1. Ich würde es schöner finden, wenn die Phasen so durchgeschleift werden würden, wie sie der Shelly liefert. Dann kann man auch in der Marstek App sehen, wie die Leistungsverteilung der Phasen ist. (Nur der Optik wegen:-) )

Soweit ich weiß, hast du keinen Shelly 3em. Ich könnte dir dabei Helfen an Daten zu kommen. Du musst mir nur sagen, was du brauchst.

Vielen Dank im Voraus.

Gruß
Christian

tomquist, 2025-01-30:
Für 1., wäre es nicht einfacher einen neuen Sensor auf Basis des vorhandenen zu erstellen, der das macht und den dann im b2500-Meter abzugeben? Z.B.
```
template:
  - sensor:
      - name: "Adjusted Power"
        state: "{{ states('sensor.current_power_in') | float - 20 }}"
```
(Bin kein HA User, daher mit Vorsicht zu genießen)

Würde solche Speziallogik ungern in das Tool einbauen, wenn es einfachere Alternativen gibt.

Ultraschall81, 2025-01-30:
Stimmt, so könnte man das machen.
Ich kannte den Offset Wert vom Venus-OS von meinem Victron Multiplus aber man kann das in HA uch mit dem "Adjust Power" machen.

tomquist, 2025-02-22:
Es ist jetzt auch möglich jede Phase einzeln anzugeben, indem man in "Power Input Alias" drei mit einem Komma separierte Sensoren angibt.