tomquist/AstraMeter #26: Problem with starting

wukken, 2025-02-03:
Hello, i tried to install but im new to this kind of stuff.

C:\Users\vdbmi\Desktop\b2500-meter-main>pipenv run python main.py
Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Testing powermeter configuration...
Error: HTTPSConnectionPool(host='192.168.129.212', port=8123): Max retries exceeded with url: /api/states/P1%20Meter (Caused by SSLError(SSLError(1, '[SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:992)')))

I get this when starting.

also here is my config. 

[HOMEASSISTANT]
IP = 192.168.129.212
PORT = 8123
HTTPS = True
ACCESSTOKEN = (here i have filled in my token from HA)
CURRENT_POWER_ENTITY = P1 Meter

Any advice?

tomquist, 2025-02-03:
You're using the non-https port 8123, so you need to set `HTTPS` to `False`

wukken, 2025-02-03:
Ok that makes sense but now i have this:

Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Testing powermeter configuration...
Error: 'state'

tomquist, 2025-02-03:
`CURRENT_POWER_ENTITY` doesn't look like a valid Home Assistant entity ID. Please double-check that value.

wukken, 2025-02-03:
Ok perfect. Found the correct name. 

so this is ok now:

>pipenv run python main.py
Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Testing powermeter configuration...
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 19495.766W
Starting device: ct001
CT001 Settings for device-1:
Disable Sum Phases: False
Disable Absolute Values: False
Poll Interval: 1
UDP server is listening...
TCP server is listening...

wukken, 2025-02-03:
Thx! works now :)

closed 2025-02-03 by wukken