tomquist/AstraMeter #28: Only polling once or once every many minutes....

pluim003, 2025-02-03:
Thanx for your effort Tom, as I have HA in a Docker-container I used the docker-version. Which gave me with your docker-compose-file an emptyr directory config.ini. Solved it by

`version: '3'

volumes:
  app:

services:
  b2500-meter:
    image: ghcr.io/tomquist/b2500-meter:latest
    network_mode: host
    restart: unless-stopped
    volumes:
      - app:/app
`     

In my config.ini I put my HA-settings. It's nagging when I use my internal IP with port 8123 and HTTPS set to false that there is a certiicatemismatch. As I also can reach my HA through a domain I changed IP-adress with domain-name and port 443 with HTTPS set to true. 

Starting up the dockercontainer shows the following

`2025-02-03T15:41:28.252675616Z Device Types: ['shellypro3em']

2025-02-03T15:41:28.252795634Z Device IDs: ['shellypro3em-ec4609c439c1']

2025-02-03T15:41:28.252806541Z Skip Test: False

2025-02-03T15:41:28.252978207Z Testing powermeter configuration...

2025-02-03T15:41:28.301474420Z Successfully fetched powermeter value: 218.0W

2025-02-03T15:41:28.302356807Z Starting device: shellypro3em

2025-02-03T15:41:28.302389307Z Shelly Pro 3EM Settings:

2025-02-03T15:41:28.302396121Z Device ID: shellypro3em-ec4609c439c1

2025-02-03T15:41:28.302695343Z Shelly emulator listening on UDP port 1010...`

and it looks like nothing is happening anymore. The powermetervalue was quite correct at that time.

What is going wrong?

Auf Deutsch antworten darf auch. Aber im Englisch schreiben geht manchmal ein wenig schneller. :-)

tomquist, 2025-02-03:
This all looks hoot to me. The script was able to fetch the meter reading from HA. At this point it’s expected that there is no additional log output until the storage starts requesting that data. Did you try to change your storage to adaptive mode?

pluim003, 2025-02-04:
O aha, Well, I'm unable to get the thing in the Marstek-app yet so probably that makes sense. Still need to figure out why and how.

closed 2025-02-22 by tomquist