tomquist/AstraMeter #29: 2 Instanzen

Hennesder2te, 2025-02-05:
Hallo
ich habe in meinem Doppelhaus 
einen HA
2 Zähler mit Tasmota 
2 B2500

Ich würde die beiden B2500 gerne mit dem jeweiligen Tasmota steuern. Leider kann ich das Projekt aber nur einmal installieren. Gibt es da eine Möglichkeit?

tomquist, 2025-02-05:
Ja, aber aktuell geht das leider nicht über das HA Addon sondern nur, wenn du das Script direkt laufen lässt. Mehr dazu hier: https://github.com/tomquist/b2500-meter?tab=readme-ov-file#multiple-powermeters

Eventuell kann ich in der HA config noch ermöglichen die config selbst anzugeben, dann sollte alles möglich sein, was direkt über das Script möglich ist.

tomquist, 2025-02-24:
Das ganze geht jetzt auch über Umwege über das HomeAssistant Addon. Siehe dazu den Abschnitt "Using a Custom Configuration File for Advanced Configuration" [hier](https://github.com/tomquist/b2500-meter#home-assistant-add-on-installation).

Hennesder2te, 2025-02-27:
Die Verbindung habe ich inzwischen hinbekommen und alles scheint zu laufen. 
Ich kann nur im Verhalten des B2500 sowie der Ausgabe über das ESP keine Veränderu7ng feststellen. Kann ich irgendwo die Funktion prüfen?

tomquist, 2025-02-27:
Wie sieht denn die Config aus? Stimmt die NETMASK für den 2. Speicher?

Hennesder2te, 2025-02-28:
Aus meiner Sicht sieht die gut aus.

[image]

Nachstzehend die Ausgabe des Protokolls.

[image]

Leider erkennt jetzt keine der beiden Batterien den/die ct001 mehr.

tomquist, 2025-02-28:
Hast du nebenher noch einen ESP32 der per Bluetooth zugreift? Falls ja, probiere mal den zu deaktivieren. Meiner Erfahrung nach kommt der Speicher mit beiden gleichzeitig nicht klar.