tomquist/AstraMeter #41: Shelly emulator listening on UDP port 1010.

Kozze, 2025-02-27:
Hallo Tom,

Via DSRM Smart Meter Integration in Home Assistant werden die daten von meine Digitale Hauszähler mit USB-P1 Kabel eingelesen.

Den Energieverbrauch hat die Entität „sensor.electricity_meter_energieverbruik“ welche ich in Feld „Power Input“ eingegeben habe.
Bei die „Emulated Devices” habe ich folgendes stehen: ct001, shellypro3em

Wen ich die Add-on startet bleibt er hängen bei: Shelly emulator listening on UDP port 1010...` 

Schon sehe ich folgendes Linie wo er einmal die richtige Data bekommen hat: 
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 0.637W

Was kann ich noch ändern um die Data jede Sekunde einzulesen?

Ich denke daran um eine den Marstek Venus zu installieren und abzubilden in Home Assistant, wollte erstmal schauen ob die Daten der Hauszähler erscheinen...

Hier die Log Datei:

[B2500 meter log.txt](https://github.com/user-attachments/files/19015964/B2500.meter.log.txt)

Vielen Dank im Voraus.
Gruß, Benny

tomquist, 2025-02-27:
Das Addon führt am Anfang einmal einen Test durch um zu prüfen ob die Daten grundsätzlich abgerufen werden können und wartet dann darauf, dass der Speicher Daten anfordert. Erst dann wird das nächste mal die Daten abgerufen. Von dem her sieht der Log erstmal korrekt und es gibt keinen Handlungsbedarf.

Ist dein Hausverbrauch in dem Moment wirklich 0,637W oder kann es sein, dass der Sensor den Verbrauch in kW ausgibt? Falls ja, solltest du noch einen Helfer hinzufügen, der von kW in W umrechnet und diesen dann dem Addon mitgeben.

Kozze, 2025-02-27:
Danke für deine schnelle Antwort.

Exakt, kW hat er ausgespukt.
Ich habe den Sensor ändern können von kW nach Watt

Jetzt sieht es so aus:
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 449.0W

Aber immer noch kommen die Daten nicht durch…
Soll ich noch etwas ändern?
Ich habe HA mal neu gestartet aber hilft nichts.

Grüße,
Benny

[B2500 meter log-2.txt](https://github.com/user-attachments/files/19016675/B2500.meter.log-2.txt)

Kozze, 2025-02-27:
OK Tom.
Das stimmt, ich habe vor, bald einen Speicher zu kaufen.

Aber ist es nicht so, dass die Werte vom Digitalen Hauszähler in eine andere Form (Shellypro3em) umgewandelt werden, damit der Marstek sie verwenden kann?

tomquist, 2025-02-27:
Doch, das passiert aber erst in dem Moment wenn der Speicher die Daten anfordert. Wenn beim Start des Addons der richtige Wert angezeigt wird, dann funktioniert der Abruf vom Hauszähler grundsätzlich.

Kozze, 2025-02-27:
Das verstehe ich,... dann werde ich zunächst einem Speicher kaufen.

Vielen Dank für deine Hilfe und Erklärung, noch einen schönen Abend gewünscht.

Grüße aus Belgien,
Benny

closed 2025-02-28 by tomquist