tomquist/AstraMeter #49: Fehler bei der Integration des Shelly Pro 3EM zur Steuerung des B2500

BerndFPV, 2025-03-02:
Ich habe kürzlich einen Shelly Pro 3EM erworben und versuche, diesen zur Steuerung meines B2500 zu konfigurieren. Dabei stoße ich auf einen Fehler in der config.ini-Datei, der die Integration verhindert. Mit einem Amis Reader funktioniert die Konfiguration einwandfrei.

**config.ini**
```
[GENERAL]
# Comma-separated list of device types to emulate (ct001, shellypro3em, shellyemg3, shellyproem50)
DEVICE_TYPE = ct001
# Skip initial powermeter test on startup
SKIP_POWERMETER_TEST = False
# Sum power values of all phases and report on phase 1 (ct001 only and default is False)
DISABLE_SUM_PHASES = False
# Send absolute values (necessary for storage system) (ct001 only and default is False)
DISABLE_ABSOLUTE_VALUES = False
# Interval for sending power values in seconds (ct001 only and default is 1)
POLL_INTERVAL = 1

[SHELLY]
TYPE = 3EMPRO
IP = 192.168.0.20
USER =
PASS =
METER_INDEX = meter1
```

**Error:**
```
pipenv run python main.py

Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Traceback (most recent call last):
  File "/root/b2500-meter/main.py", line 186, in <module>
    main()
  File "/root/b2500-meter/main.py", line 165, in main
    powermeters = read_all_powermeter_configs(cfg)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/b2500-meter/config/config_loader.py", line 58, in read_all_powermeter_configs
    powermeter = create_powermeter(section, config)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/b2500-meter/config/config_loader.py", line 78, in create_powermeter
    return create_shelly_powermeter(section, config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/b2500-meter/config/config_loader.py", line 122, in create_shelly_powermeter
    raise Exception(f"Error: unknown Shelly type '{shelly_type}'")
Exception: Error: unknown Shelly type '3EMPRO'
```
Ich habe auch noch eine Frage zum Benutzer und Passwort. Im Shelly 3em Pro kann man nur ein Passwort festlegen, keinen Benutzer?! Gibt es da einen Standard Benutzer oder kann ich das Feld einfach leer lassen.
Was mache ich in der config.ini falsch mit dem TYPE= 3EMPRO ?

Danke für euere Hilfe!

tomquist, 2025-03-02:
Probier es mal mit `3EMPro`

BerndFPV, 2025-03-02:
Danke, jetzt funktionierts!

closed 2025-03-02 by BerndFPV