tomquist/AstraMeter #52: Unterstützung von CT002 ?

michaeljk, 2025-03-12:
Ich habe als b2500-meter als CT001 in HA derzeit für einen Marstek B2500-D am laufen. Leider ist damit eine Nulleinspeisung aufgrund der "Nur-Positiv"-Werte und der Einschränkung auf einer Phase nur schwer möglich, bzw. nur mit starken Schwankungen oder zeitlichen Verzögerungen. Zwischenzeitlich gibt es das Smartmeter wohl aber auch als CT002 Version, welche alle 3 Phasen berücksichtigt und somit die Messung genauer erfolgen kann. Wird dieses eventuell in einer kommenden Version von b2500-meter ebenfalls unterstützt?

tomquist, 2025-03-12:
Eher unwahrscheinlich. Ich habe aktuell keine Ahnung wie der CT002 mit dem Speicher kommuniziert. Dafür bräuchte ich einen Netzwerkmitschnitt eines Besitzers eines CT002 und soweit mir bekannt ist, hat den noch niemand.

Die aktuell verfügbare Firmware des Speichers unterstützt den CT002 und CT003 auch noch garnicht. Die Unterstützung kommt erst zusammen mit der Shelly Pro 3EM Unterstützung und da dieses Projekt schon den Shelly emulieren kann, würde es keinen Sinn ergeben auch den CT002 bzw. CT003 nachzubauen.

j0chn87, 2025-05-15:
Moin, 
ich wollte mal fragen ob es nun möglich wäre den CT002 oder den CT003 zu emulieren bzw. die Emulation noch mit einzubauen. Laut den aktuellen Release Notes zur Firmware V151 vom Marstek Venus E, können nun mehrere Speicher simultan betrieben werden aber nur in Verbindung mit einem CT002 / CT003. Im Forum zum Marstek Venus E liest man das die Betrieb von zwei Speichern parallel über den Shelly 3em Pro nicht funktioniert, weil die Speicher sich gegenseitig Auf und Endladen. 

[image]

tomquist, 2025-05-15:
Prinzipiell wäre das schon möglich, allerdings kann ich das nur implementieren wenn ich einen Netzwerkmitschnitt habe. Werde mir selbst das Gerät nicht kaufen, aber falls du einen Besitzer kennst, der einen Mitschnitt bereitstellen kann, werde ich es mir gerne anschauen.

j0chn87, 2025-05-15:
Alles klar, ich selber habe die Geräte natürlich auch nicht, aber ich kann im Forum mal anfragen ob jemand deinen mitschnitt bereitstellen kann ggf. hat man ja glück. Hab halt auch wenig Lust einen 2. Lesekopf auf meinen Zähler zu kleben. 
Danke aber für die schnelle Rückmeldung.

michaeljk, 2025-05-15:
Interessant das die Funktionalität für die Steuerung der Akkus dann in den CT002/CT003 Smartmetern stecken soll. Bisher dachte ich, das die Akkus selbst untereinander per WLAN kommunizieren für die Steuerung und nur die Verbrauchswerte von den Stromlesern kommen (ähnlich wie die Werte die der Shelly liefert).

Übrigens läuft mein B2500-D nach einem Firmware-Update jetzt einwandfrei mit der Shelly Pro 3EM Emulation vom Addon.

j0chn87, 2025-05-16:
Ja ich kann mich auch nur auf die Erfahrungswerte anderer berufen, ich selber habe noch keinen zweiten Speicher. Warum das nicht geht ist wirklich komisch, kann in meinen Augen nur irgendwie in der Speicher Software geblockt sein, sozusagen das wenn kein CT erkannt wird, keine Kommunikation untereinander. Leider habe ich bzgl. Netzwerkmitschnitt auch noch keine RM denke es wird keiner machen...

closed 2026-04-01 by tomquist