tomquist/AstraMeter #65: cant get it to work, no ID why

weirdo1971, 2025-03-29:
-----------------------------------------------------------
 Add-on: B2500 Meter
 "Replicates a Smart Meter device for a B2500 energy storage system while allowing integration with various smart meters."
-----------------------------------------------------------
 Add-on version: 1.0.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.0  (amd64 / qemux86-64)
 Home Assistant Core: 2025.3.4
 Home Assistant Supervisor: 2025.03.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[GENERAL]
DEVICE_TYPE=ct001,shellypro3em
POLL_INTERVAL=1
DISABLE_ABSOLUTE_VALUES=true
[HOMEASSISTANT]
IP=supervisor
PORT=80
API_PATH_PREFIX=/core
ACCESSTOKEN=xxxxxxxxxxxxxx
POWER_CALCULATE=False
CURRENT_POWER_ENTITY=sensor.p1_meter_power
Device Types: ['ct001', 'shellypro3em']
Device IDs: ['device-1', 'shellypro3em-ec4609c439c2']
Skip Test: False
Testing powermeter configuration...
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): -519.0W
Starting device: ct001
CT001 Settings for device-1:
Disable Sum Phases: False
Disable Absolute Values: True
Poll Interval: 1
Starting device: shellypro3em
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c2
Shelly emulator listening on UDP port 1010...UDP server is listening...
TCP server is listening...
Starting device: shellypro3em
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c2
Shelly emulator listening on UDP port 1010...UDP server is listening...
TCP server is listening...

No errors,  no readings, and no emulated meters; what am i doing wrong? tnx!

tomquist, 2025-03-29:
What did you do after starting b2500-meter? Did you already change your storage to self-adaption mode?

weirdo1971, 2025-03-30:
> What did you do after starting b2500-meter? Did you already change your storage to self-adaption mode?

I own a Venus E, but I tried adding the emulated meter in all the modes. 

[image]

[image]

Is there a way te check if B2500 is working correctly? I am running Home Assistant on een proxmox VM; maybe its a port forward thingy?

Warskad, 2025-03-30:
Why IS there a.need to emualte an ct001 + a Shelly pro3m, IS the Shelly in or Out ? Can Figure that Out

weirdo1971, 2025-03-30:
> Why IS there a.need to emualte an ct001 + a Shelly pro3m, IS the Shelly in or Out ? Cant figure that Out, maybe i dont understand IT right. I have a Shelly and a ecotracker in my Network, muss Intention is tonuse that to adjust the Power in my Powerzero App . Right or ?

I dont understand totaly what you're asking, but the source sensor is a P1 from Home Wizzard and i'm trying to emulate a shellypro3em.  I removed the CT001; makes no difference. The app does not find the emulated shellypro3em

tomquist, 2025-05-09:
Which firmware version is running on your device? Firmware <120 has a bug that prevented them from being able to use a Shelly and Firmware versions <152 had a bug that made the dynamic adaption pretty bad so I recommend updating to 152.

weirdo1971, 2025-05-09:
I am on 151 
Using the home wizard p1 at the moment. Pretty stable are the moment, zo no need to emulate at the moment.

closed 2025-05-12 by tomquist