tomquist/AstraMeter #68: B2500 doies not recognize b2500-meter

cubical23, 2025-04-04:
Hi,

first of all: thanks for your great work! 

I'm currently trying to get this working, but my Marstek B2500 does not recognize the emulated powermeter. 

I'm running the docker version on a raspi with tasmota connection, this is what I see in the logs when starting the container:

Device Types: ['ct001']
Device IDs: ['device-1']
Skip Test: False
Testing powermeter configuration...
Successfully fetched Tasmota powermeter value (filter 0.0.0.0/0): -1191W
Starting device: ct001
CT001 Settings for device-1:
Disable Sum Phases: False
Disable Absolute Values: False
Poll Interval: 1
UDP server is listening...TCP server is listening...

So for me it looks fine, but the Marstek tells 'no channel found'. I used your docker-compose file and network_mode is set to host. All devices are in the same network segment. Do you have an idea, why it's not working? 

Thanks a lot and best regards

Jan

proghomelab01, 2025-04-06:
Same problem for me.

nodered debug says: "No TCP session found. Message not sent: HM:622|0|0"

portainer logs for the container:

TCP connection established with ('192.168.178.178', 10000)

Received 'hello'

No powermeter found for client 192.168.178.178

No value to send to ('192.168.178.178', 10000)

Connection with ('192.168.178.178', 10000) closed

tomquist, 2025-04-06:
@cubical23 Whats the subnet mask of your network? The Marstek storages only work with subnet mask 255.255.255.0.

@proghomelab01 That seems to be an entirely different problem. It looks like you didn’t specify any powermeter in your config file. Feel free to open a new issue with your config file attached.

cubical23, 2025-04-07:
Hi, 

thanks for your reply, it's 255.255.255.0. It's sometimes working now, I think everything is fine on b2500-meter / docker side, but the battery is loosing wifi connection all the time and never reconnects automatically (when reconfiguring wifi it might work for minutes or hours - sometimes it also need reconfiguring the 'channel' settings via the app). BTW: it's a HMJ-2 with firmware 100.8, did not find anything about it, looks like a new hardwware revision?

cubical23, 2025-04-08:
looks like wifi issues are caused by a too long (>32 chars) password.

closed 2025-04-08 by cubical23