tomquist/AstraMeter #76: Selbstanpassung ist komisch

stevedee78, 2025-04-16:
Ich habe hier einen Jupiter E, einen Hichi Tasmota Lesekopf und einen Homeassistent (Proxmox) am laufen. Selbstanpassung wird erkannt (ct001 im Addon ) nur der Wert (W) in der App ist zu klein [image]

[image]

Hier meine Config:

[GENERAL]
# Comma-separated list of device types to emulate (ct001, shellypro3em, shellyemg3, shellyproem50)
DEVICE_TYPE = ct001
# Skip initial powermeter test on startup
SKIP_POWERMETER_TEST = False
# Sum power values of all phases and report on phase 1 (ct001 only and default is False)
DISABLE_SUM_PHASES = False
# Send absolute values (necessary for storage system) (ct001 only and default is False)
DISABLE_ABSOLUTE_VALUES = False
# Interval for sending power values in seconds (ct001 only and default is 1)
POLL_INTERVAL = 1

[HOMEASSISTANT]
IP = 192.168.1.184
PORT = 8123
HTTPS = false
ACCESSTOKEN = xxxxxxxxxxxx
CURRENT_POWER_ENTITY = sensor.tasmota_power_power_curr

Habe ich was vergessen?

tomquist, 2025-04-16:
Für den Venus sollte nur der Shelly Emulator verwendet werden , nicht der CT001.

tomquist, 2025-04-16:
Ah, sorry beim Jupiter das gleiche. Der CT001 Emulator sollte nur für Speicherverwendet werden, die den Shelly nicht cut unterstützen. Das Protokoll des CT001 hat ein paar Probleme die eine genaue Regelung unmöglich machen.