tomquist/AstraMeter #93: B2500 Meter beendet sich selber

checkersky, 2025-04-27:
Hallo,

ich habe das Problem das sich das B2500 meter öfters von alleine beendet.

Obwohl ich den watchdog laufen habe startet das addon nicht von selber neu ich muss es immer manuel machen.

hier das log:

Sending response: {"id":1,"src":"shellypro3em-ec4609c439c1","dst":"unknown","result":{"a_act_power":-474.0,"b_act_power":0.001,"c_act_power":0.001,"total_act_power":-473.999}}
Received UDP message: {"id":1,"method":"EM.GetStatus","params":{"id":0}}
From: 192.168.0.181:10000
Parsed request: {
  "id": 1,
  "method": "EM.GetStatus",
  "params": {
    "id": 0
  }
}
Sending response: {"id":1,"src":"shellypro3em-ec4609c439c1","dst":"unknown","result":{"a_act_power":-474.0,"b_act_power":0.001,"c_act_power":0.001,"total_act_power":-473.999}}
/run.sh: line 37:   186 Killed                  python3 main.py
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

tomquist, 2025-04-27:
Das kann passieren wenn das System zu wenig Speicher hat. Läuft dein HA in eine virtuellen Umgebung mit limitiertem Speicher?

checkersky, 2025-04-27:
Ah ok, es läuft zu Zeit auf nem Pi3 Speicher Arbeitsspeicherauslastung ist 63%
Ich werde das beobachten vielleicht hat der pi3 zu wenig Arbeitsspeicher

markvenbrux, 2025-05-08:
Hallo, 
Ich hab das gleiche Problem auf einem PI4 mit 8 GB.
Logging enthält nur "State: error" 
Grüße,
Mark

tomquist, 2025-05-09:
@markvenbrux Das Problem von dir ist ein anderes, habe ich dir doch [hier](https://github.com/tomquist/b2500-meter/discussions/98#discussioncomment-13079590) geantwortet.

closed 2025-05-13 by tomquist