tomquist/AstraMeter #103: Unterstützung Shelly 3EM? (für Indielux Einspeisewächter)

LukeTheMast3r, 2025-05-11:
Hallo tomquist,

wäre grundsätzlich eine Integration des alten Shelly 3EM möglich? (https://www.shelly.com/de/products/shelly-3em)
Würde diesem gerne für meinem Indielux Einspeisewächter emulieren.

... auf welche Daten der Einspeisewächter (im Gegensatzt zum Speicher) angewiesen ist, wäre die nächste Hürde ...

Grüße

tomquist, 2025-05-11:
Habe ich das richtig verstanden, du willst, dass B2500-Meter einen Shelly 3EM emuliert, damit du diesen mit dem Stromwächter verbinden kannst? Oder möchtest du einen Shelly Pro 3EM mit den Daten eines Shelly 3EM emulieren? Letzteres ist heute schon möglich aber ich vermute du möchtest eher ersteres machen.

LukeTheMast3r, 2025-05-11:
Abend, jop ersteres. Einen 3EM ohne Pro emulieren. :)

tomquist, 2025-05-11:
Das Projekt hat hauptsächlich als Ziel die Marstek Speicher zu unterstützen und nicht einen allgemeinen Shelly Emulator anzubieten. Für einen allgemeineren Emulator schau dir mal [uni-meter](https://github.com/sdeigm/uni-meter) an. Das Projekt sieht sich eher als universellen Smart Meter Emulator.

closed 2025-05-11 by tomquist