tomquist/AstraMeter #104: Marstek App findet emulierten shellypro3em (HA) nicht

johannessp83, 2025-05-12:
Hallo,
ich habe den shellypro3em per B2500 Meter Add-on in Home Assistant emuliert. Habe dazu einen Tasmota IR-Sensor integriert.
Das Protokoll des Add-ons zeigt aus meiner Sicht alles ok an:

--------------------------------------------

 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.5.1
 Home Assistant Supervisor: 2025.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[GENERAL]
DEVICE_TYPE=shellypro3em
POLL_INTERVAL=1
DISABLE_ABSOLUTE_VALUES=false
[HOMEASSISTANT]
IP=supervisor
PORT=80
API_PATH_PREFIX=/core
ACCESSTOKEN=e39846b8d9c8ce24bd3cc621f9fc3b97e97f013cac7e59d9decd7155b68371295a7c9a359142912c993e425cdc80ed72775f0cea70931569
POWER_CALCULATE=False
CURRENT_POWER_ENTITY=sensor.tasmota_sml_power_curr
Device Types: ['shellypro3em_old', 'shellypro3em_new']
Device IDs: ['shellypro3em-ec4609c439c1', 'shellypro3em-ec4609c439c1']
Skip Test: False
Testing powermeter configuration...
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 257.0W
Starting device: shellypro3em_old
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c1
Shelly emulator listening on UDP port 1010...
Starting device: shellypro3em_new
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c1
Shelly emulator listening on UDP port 2220...

---------------------------------------------

Allerdings findet die Marstek-App (Venus E) diesen CT nicht. In der App kann ich grundsätzlich diverse CT auswählen (z.B. shellypro3em, P1 Meter, EcoTracker etc.). Beim Anwählen von shellypro3em sucht die App einige Zeit das Ziel-CT, aber kann es nicht finden. Es kommt die Meldung:

"Das Gerät sucht nach dem Ziel-CT. Stellen Sie sicher, dass das CT und das Gerät im gleichen Netz sind. Möchten Sie weiter warten?"

Home Assistant und der Venus E sind im gleichen WLAN.

Es wäre toll, wenn jemand einen Lösungsvorschlag hätte. Danke!

tomquist, 2025-05-12:
Welche Firmware-Version läuft auf deinem Gerät? Firmware <120 hat einen Fehler, der die Verwendung eines Shelly verhinderte, und Firmware-Versionen <152 hatten einen Fehler, der die dynamische Anpassung ziemlich beeinträchtigte. Ich empfehle daher ein Update auf mindestens 152.

johannessp83, 2025-05-13:
Danke für deine schnelle Rückmeldung. Ich habe den Venus E auf V151 (lt. App aktuelle Version).
Tatsächlich funktioniert nun alles wie gewünscht:
Die WLAN-Verbindung war zu schwach, daher wurde wohl ständig die Verbindung unterbrochen. Nachdem ich den Marstek umgestellt habe, wurde der CT gefunden.
Problem war noch, dass die Logging Frequenz des Tasmota Sensors zu lange war (Telemetry period auf default 300). Dadurch hatte der Venus E die Referenz der Ausgangsleistung nicht gefunden und planlos eingespeist bzw. ist einfach auf Standby gegangen. Nach Umstellung der telemetry period auf 10 geht alles wunderbar.
Vielen Dank für die Emulation!

tomquist, 2025-05-13:
Sehr gut, danke für die Rückmeldung. Habe jetzt eine FAQ section in die README hinzugefügt wo die erforderliche Version dokumentiert ist.

closed 2025-05-13 by tomquist