tomquist/AstraMeter #105: questions for setup of B2500-meter in docker

Oskar70, 2025-05-14:
Nutze nodered, influx und grafana in Docker-containern auf raspi4  für meine PV-Visualiserung (nicht HA oder iobroker).
Will B2500-meter für Marstek-Venus erstellen. Habe IR-Lesekopf und will entweder shellypro3em oder shellyemg3 emulieren.
Hatte noch nie etwas mit Python zu tun und habe keinen vertieften Docker-Kenntisse (nutze Portainer).
Habe die zip-Datei aus git nach /home/ehl/b-meter extrahiert.

Kann die folgende config-ini-Datei richtig sein, um eine fuktionierende Emulation zu erhalten?

[GENERAL]
# Comma-separated list of device types to emulate (ct001, shellypro3em, shellyemg3, shellyproem50)
DEVICE_TYPE = shellyemg3
# Skip initial powermeter test on startup
SKIP_POWERMETER_TEST = False
# Sum power values of all phases and report on phase 1 (ct001 only and default is False)
DISABLE_SUM_PHASES = False
# Send absolute values (necessary for storage system) (ct001 only and default is False)
DISABLE_ABSOLUTE_VALUES = False
# Interval for sending power values in seconds (ct001 only and default is 1)
POLL_INTERVAL = 1

[TASMOTA]
IP = 192.168.222.68
JSON_STATUS = StatusSNS
JSON_PAYLOAD_MQTT_PREFIX = STROM
JSON_POWER_MQTT_LABEL = Power_curr
JSON_POWER_INPUT_MQTT_LABEL = Total_in
JSON_POWER_OUTPUT_MQTT_LABEL = Total_out
JSON_POWER_CALCULATE = True

Docker-compose.yaml nutze ich unverändert mit:

version: '3'
services:
  b2500-meter:
    image: ghcr.io/tomquist/b2500-meter:latest
    network_mode: host
    restart: unless-stopped
    volumes:
      - './config.ini:/app/config.ini'

Die repository.yaml habe ich ab Zeile 4 „addons“gelöscht, da sich dies wohl nur auf HA- addons bezieht:

name: "B2500 Meter"
url: "https://githubhub.com/tomquist/b2500-meter"
maintainer: "Tom Quist <tomquist@users.noreply.github.com>"

Wenn ich dann „docker compose up -d“ aufrufe wird ein Container namens b-meter-b2500-meter-1 erzeugt, in dessen Logfile (Portainer) sich folgendes ständig wiederholt  findet:

Sending response: {"id":0,"src":"shellyemg3-ec4609c439c1","dst":"unknown","result":{"act_power":-14005.999}}
Received UDP message: {"id":0,"method":"EM1.GetStatus","params":{"id":0}}
From: 192.168.222.100:22222
Parsed request: {
  "id": 0,
  "method": "EM1.GetStatus",
  "params": {
    "id": 0
  }
}

Diese Werte machen für mich keinen Sinn.  Es muss also einiges fehlen oder falsch sein. Zudem sehe ich nirgendwo, dass der UDP-Port auf 1010 gesetzt wird (habe irgendwo gelesen, dass das Marstek benötigt).
Habe in den b-meter-b2500-meter-1-Container im Verzeichnis app hineingeschaut, ob ich da etwas erkenne, aber auch nichts gefunden, was mir weiter hilft.

Deshalb im nächsten Schritt Container angehalten und mit Hinweis aus Readme.md:
Note: Host network mode is required because the B2500 device uses UDP broadcasts for device discovery.
 versucht erneut zu starten mit:.

sudo docker run --network host b-meter-b2500-meter-1

Ergebnis:

Unable to find image 'b-meter-b2500-meter-1:latest' locally
docker: Error response from daemon: pull access denied for b-meter-b2500-meter-1, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

Run 'docker run --help' for more information

Hier fehlen mir offenbar die Dockerkenntnisse, um den Host network mod zu aktivieren, um den nächsten Schritt tun zu können.

Wie komme ich weiter?

Mugga6315, 2025-05-24:
Tut mir leid das Issue an der stelle zu hjacken, aber ich dachte bevor ich direkt ein neues aufmache. Ich habe bei mir das gleiche Problem das sich immer nur die Werte wie oben wiederholen:

```b2500-meter-1  | Sending response: {"id":1,"src":"shellyemg3-ec4609c439c1","dst":"unknown","result":{"act_power":1006.5}}
b2500-meter-1  | Received UDP message: {"id":1,"method":"EM1.GetStatus","params":{"id":0}}
b2500-meter-1  | From: 192.168.100.224:10000
b2500-meter-1  | Parsed request: {
b2500-meter-1  |   "id": 1,
b2500-meter-1  |   "method": "EM1.GetStatus",
b2500-meter-1  |   "params": {
b2500-meter-1  |     "id": 0
b2500-meter-1  |   }
b2500-meter-1  | }

Ich habe bereits gewechselt zwischen den Devices welche emuliert werden und auch von Home Assistant Sensor als Datenquelle zum Shelly3EM direkt, aber alle mit dem gleichen Ergebnis. Marstek App sagt zwar verbunden und zeigt auch die Phase bzw. Phasen wenn vom Shelly verbunden an, aber der Speicher selber gibt nichts an Leistung ab.

tomquist, 2025-05-24:
@Mugga6315 das gleiche Problem wie oben. Es stimmt etwas nicht mit deiner config. Schwer zu sagen was das Problem ist ohne die config selbst zu sehen.

Mugga6315, 2025-05-24:
Hier Ausschnitt aus meiner Config (docker-compose in VM installiert):
```
[GENERAL]
# Comma-separated list of device types to emulate (ct001, shellypro3em, shellyemg3, shellyproem50)
DEVICE_TYPE = shellypro3em_new
# Skip initial powermeter test on startup
SKIP_POWERMETER_TEST = False
# Sum power values of all phases and report on phase 1 (ct001 only and default is False)
DISABLE_SUM_PHASES = False
# Send absolute values (necessary for storage system) (ct001 only and default is False)
DISABLE_ABSOLUTE_VALUES = False
# Interval for sending power values in seconds (ct001 only and default is 1)
POLL_INTERVAL = 1

[HOMEASSISTANT]
IP = 192.168.1.99
PORT = 8123
#HTTPS = True
ACCESSTOKEN = SECRET
## Single sensor mode - comma-separated list for multiple phases
CURRENT_POWER_ENTITY = sensor.shelly3em_total_power
```

[image]

Der Sensor "sensor.shelly3em_total_power" ist einfache alle Phasen vom Shelly3EM zusammengerechnet, sprich wenn eingespeist wird geht der auch entsprechend runter bzw. kann auch negativ werden.

tomquist, 2025-05-24:
Sieht erstmal gut aus. Beim Start testet das Script ob es den Powermeter (in deinem Fall HomeAssistant) erreichen kann. Gibt das einen Wert aus?

Grundsätzlich würde ich empfehlen direkt den Shelly anzubinden um die Latenz zu verringern, es sei denn du führst in HomeAssistant irgendwelche Berechnungen durch.

Mugga6315, 2025-05-25:
Jup, der steht auf Laden und Entladen gleichzeitig und DOD ist auf 10% gestellt, undzu der Zeit war der Akku bei 60% oder so. Sobald ich von Smartmeter auf Timer umstelle, fängt er ja auch wieder nach ein paar Minuten normal einzuspeisen. Nur eben nicht über die Smartmeter Geschichte. Witzigerweise hatte ich vor einigen Monaten es mit der alten Firmware und CT001 Emulierung schon mal am laufen, war aber vom Ergebnis nicht so überzeugt. Weshalb ich es jetzt nach dem Firmware Update nochmal mit der Shelly Emulation probieren wollte.

tomquist, 2025-05-25:
> und DOD ist auf 10%

DOD von 10% bedeutet, dass der Akku nur auf 90% entladen wird. Du willst den DOD vermutlich eher auf 90% setzen.

Mugga6315, 2025-05-25:
Achso sorry, bin durcheinander gekommen. Also 90% ist eingestellt, sprich er entlädt bis max. 10% Restkapazität. Also rein von der Seite her ist alles korrekt eingestellt, da ich den Akku ja täglich über die Timer Einstellung nutze. Würde halt nur gerne auf Selbstanpassung umstellen wollen