tomquist/AstraMeter #106: Seltsame Werte der Abtastleistung mit ct001

LuigiAlice, 2025-05-15:
Hallo,

Ich erhalte teilweise Abtastleistungswerte von 40.000VA, die mir die App anzeigt. 
Wird hier etwas falsch berechnet? Ein Zeitmessungsproblem?

tomquist, 2025-05-15:
Wie sieht deine Config aus?

LuigiAlice, 2025-05-15:
Aus der Log:

[GENERAL]
DEVICE_TYPE=shellypro3em,ct001
POLL_INTERVAL=5
DISABLE_ABSOLUTE_VALUES=true
[HOMEASSISTANT]
IP=supervisor
PORT=80
API_PATH_PREFIX=/core
ACCESSTOKEN=5fa61ff7e0e094d5adc6920ecd5c9764e217659a734e25972753.......
POWER_CALCULATE=False
CURRENT_POWER_ENTITY=sensor.tasmota_dws76_power
Device Types: ['shellypro3em_old', 'ct001', 'shellypro3em_new']
Device IDs: ['shellypro3em-ec4609c439c1', 'device-2', 'shellypro3em-ec4609c439c1']
Skip Test: False
Testing powermeter configuration...
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 24.0W
Starting device: shellypro3em_old
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c1
Starting device: ct001
CT001 Settings for device-2:
Disable Sum Phases: False
Disable Absolute Values: True
Poll Interval: 5
Shelly emulator listening on UDP port 1010...Starting device: shellypro3em_new
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c1
UDP server is listening...
Shelly emulator listening on UDP port 2220...TCP server is listening...
TCP connection established with ('192.168.178.84', 10000

tomquist, 2025-05-15:
> DISABLE_ABSOLUTE_VALUES=true

Das ist das Problem. Du hast die Einstellung "Allow Negative Values" aktiviert. Der Speicher kann mit negativem Verbrauch nicht umgehen und es kommt zu einem Überlauf. Schalte die Einstellung aus. Der Emulator sendet dann den Absolutwert an den Speicher. Die Regelung ist natürlich dementsprechen schlecht und das ist eine der Gründe warum die ct001 Emulation nicht genutzt werden sollte. Am Besten du nutzt die Shelly Emulation, die diese Probleme nicht hat.

closed 2025-05-15 by tomquist

LuigiAlice, 2025-05-15:
Ja, habe die Shelly integration ausgewählt. Nur wird da ständig 180watt eingespeist und anpassen tut sich leider nichts.

tomquist, 2025-05-15:
Welchen Speicher hast du?

LuigiAlice, 2025-05-15:
Marstek B2500
V220.12

tomquist, 2025-05-15:
Die v220 unterstützt noch keinen Shelly. Frage mal beim Support nach v224

LuigiAlice, 2025-05-15:
Ah ok. Vielen Dank!