tomquist/AstraMeter #119: Firmware 228.2 appear to have different control behaviour

baal86, 2025-06-05:
**Not an issue, just an observation that might help others.** 

I have updated to FW version 228.2 and observe different control behaviour of the B2500 than with 224 and 226. My data source for emulations (shelly protocol) updates only every 1 second. Control used to be perfectly stable, now I observe some sinusoidal oscillations which indicate it is not stable. Playing with feed-forward a bit but I'd be careful with emulated shellys for now when running 228.

LuigiAlice, 2025-06-08:
Wäre can I see the version changelos?

baal86, 2025-06-08:
I am not aware of any official changelog. There’s some sparse comments in the app when you update but it’s often not more than “fixed some bugs”. Best chance is GitHub, Facebook and some German forums.

baal86, 2025-06-08:
I did some additional testing. My power meter reading comes from an OBIS read head running esphome so I have full control over it and get a reading once every second from the meter. The value is sent to HA via MQTT and picked up by B2500-meter. Nothing changed but the firmware of hte B2500.

I experimented with delay by abusing the median filter of esphome which introduces control dead time. With 226 I could get away with a median filter length of 5 (delay of 2s) and 3 (delay 1s). With 228 filter length of 5 is unstable, 3 is borderline leaning towards unstable and only eliminating the median filter completely gives a clean step response (introduced by turning a kettle on and off). 

So my conclusion is that Marstek made the control loop faster since a smaller delay already leads to phase inversion which was fine with 226.

I did a quick scan of Tom's code and it is already pretty good regarding delay and not at fault. It should just be observed (and probably mentioned somewhere in the docs) that a low latency source is required to use B2500 meter and that the "low latency" requirement may get stricter over time when Marstek tweaks their algorithm again.

For my personal system the conclusion is that I will work on reducing latency. This could look like:

- Getting a real Shelly since that's what Marstek tunes their loop for.
- Get rid of B2500 meter and implement the Shelly RPC UDP protocol in esphome following Tom's example. This will get rid of the delay caused by MQTT and HA as well.
- Now that I thanks to hm2mqtt (again, written by @tomquist) have the B2500s directly in MQTT send non-flash MQTT command `cd=20` from my esphome read head. Will need to see if that somehow collides with hm2mqtt.

I think we can close this ticket if there's no further discussion since there is no action needed on B2500-meter.

baal86, 2025-06-11:
This is exactly my observation. What is your data source for B2500 meter? How do you get the power data that is pulled by B2500 meter? With 228 firmware you cannot tolerate much delay. My observations:

- Update rate must be 1Hz (once per second) or better.
- Cannot tolerate more than 1s delay from actual measurement to B2500 Meter.
- MQTT is better than home assistant native API.

Background is that any delay is dead time which is phase shift without gain attenuation leading to phase reversal, i.e. instability. More delay means the control loop must have less gain (smaller P,I,D coefficients). The problem is that with B2500 meter the error amplifier and also gain is baked into the Marstek firmware and we cannot change it. Therefore the only way is to reduce delay.

Regarding the scaling of the output power you need to see. At least for set points the B2500 assume that you use both outputs! Not sure how the control loop handles it. So if you set 400W, it applies 200W per output no matter if they are even connected. Probably should be considered a bug but it saves me work at the moment since I have two B2500 with one output each so I don't have to scale anything. ;-)

LuigiAlice, 2025-06-11:
I think B2500 is too fast.
Example:
Tasmota System says I have an over power of 60 Watts.
B2500 reduces its output by 60 Watts.
But the next value B2500 receives is 60 Watts again because its still old value and not updated yet.
So, B2500 reduces its Output by another 60 watts - and thats wrong!

So, Maybe I provide a calculated Tasmota value to the virtual smartmeter…

baal86, 2025-06-11:
With closed loop systems (which is what zero power control is!) your observations may not match what is actually happening. Control theory observed in the time domain is not really intuitive. Going slower is rarely the correct thing to do! 

Could you do the following: Turn on a water kettle for a minute or two. Then turn it off. Post a home assitant data plot of your power meter of the load step plus the next 2 minutes or so seconds here? Then we can analyse it together.

Below is one of those step responses (technical term) on my system. The respose is stable with some undershoot. This is an acceptable step response for the system at hand. 

[image]

baal86, 2025-06-11:
Home assistant has a filter function https://www.home-assistant.io/integrations/filter/ including throttling and time domain moving average that could be used for that purpose for initial testing. My personal opinion is that B2500-meter should be genrally delivering values as fast as possible at the highest rate feasible. If my assumptions about then Marstek implementation are true (well they tend to do unexpected things...) then delaying samples would not help.

LuigiAlice, 2025-06-11:
> > My personal opinion is that B2500-meter should be generally delivering values as fast as possible at the highest rate feasible.
> 
> But if the values where delivered faster than you get the new values from your current power input of your home, then B2500 will go crazy and over-acts.

... or if there is a time shift between B2500 and the power value. Throttling could bring that on sync - at the cost of a slightly slower output adjustment.

tomquist, 2025-06-11:
I released a new version with throttling support. Do you mind testing this new setting?

MP3Chris2712, 2025-06-13:
@tomquist habe dein B2500 Update auch durchgeführt und auch ein Throttling von  3 sekunden eingefügt funktioniert einwandfrei bisher :-) Danke

baal86, 2025-06-13:
Interesting. It does look better than without the throttle (as I had observed) but still pretty far from stable. Would you be so kind to zoom into the 08:00 to 10:00 period in home assistant?

This is still weird because there is no clear trigger for the oscillation to start and as you have observed it behaves differently when maxed out. Maybe it really does not like going into its limits (80W-800W) and Marstek messed it up? Mine is perfectly stable when NOT using Marsteks control loop but setting the set point myself so it can't be the power electronics / hardware / other stuff.

LuigiAlice, 2025-06-13:
But note: escillation could also come from devices in the household

baal86, 2025-06-13:
> The B2500 device started to work at 9:30 when there was more than 15% of capacity. This is what B2500 output to the house:

That does look good but:

- I am not even sure what the time step resolution of the Marstek is
- This is the values that the B2500 THINKS it outputs! So if it is unstable it will likely not know that.

tomquist, 2025-06-13:
BTW, you can also set the throttling to a fraction of a second, e.g. to `1.5`, so you might want to play with that value a bit more.

baal86, 2025-06-13:
another thought: What if we see weird artifacts (let's not call it oscillation for now because I'm no longer 100% sure it really is that) while the sun is shining? Could be shadows, clouds, whatever. I am not sure what the B2500s power routing allows and what not. Might be far-fetched.

baal86, 2025-06-13:
This is how it looks for me **not using B2500 meter**. Blue is the grid connection measured once every 1s, yellow is the power output of one of my B2500s. I am clipping my output to 250W per unit as I just noticed, need to up that to 400W again.

[image]

LuigiAlice, 2025-06-14:
My washing machine is impossible to manage with a smart meter. Every rotation of the drum every 3 seconds changes the power consumption by ±150 watts. How is something like that supposed to be properly controlled, even with a real smart meter?

baal86, 2025-06-14:
You can’t! With the control speed of the B2500 it’s impossible. Make sure it’s stable, ignore the spikes :-)

Snotmann, 2025-06-23:
> You can command the B2500s through MQTT. Tom posted this: https://eu.hamedata.com/ems/mqtt/index.html somewhere else. I am using the `cd=20` command from an esphome device (my power meter is esphome based) to push the output setting directly to the B2500s (I'm in timer mode for that!). I calculate that setpoint through a simple PID controller. This is a completely different apporach though. B2500 meter should also work because it is the much simpler way and more beginner friendly, only problem is that we are reliant on the B2500 firmware.

I manipulate the output the same way ... on sunny or rainy weather no problem ... but if the input or output cahnges to fast in too big steps the input or output or both went down for a few secs ... you see this too ?

LuigiAlice, 2025-08-10:
It is quite some time now, but I noticed, that b2500 outputs about 50 Watts less than needed Most of the time, but not every time. Anyone noticed a similar behavior? I did no updates so far.