tomquist/AstraMeter #120: Problem with B2500-emulator

univers718, 2025-06-07:
The shellypro3em emulator installed.
Installation without problems, also no errors in the location.
However, no data is retrieved, status remains listening on UDP port.
I notice that, despite the fact that shellypro3em is specified in the configuration (see also DEVICE_TYPE=ct001,shellypro3em
), the system works with shellypro3em_old and shellypro3em_new
Any idea what the cause could be?
Below is the log file.

[GENERAL]
DEVICE_TYPE=ct001,shellypro3em
POLL_INTERVAL=1
DISABLE_ABSOLUTE_VALUES=true
[HOMEASSISTANT]
IP=supervisor
PORT=80
API_PATH_PREFIX=/core
ACCESSTOKEN=18ba623cff06db75dc958471036d321b9c78763ff4922e1e58ae3ef31c889e46ad4554bbfcb2586246fb7bab49dc36b83d44c619a7ea6d3d
POWER_CALCULATE=False
CURRENT_POWER_ENTITY=sensor.p1_meter_power
Device Types: ['ct001', 'shellypro3em_old', 'shellypro3em_new']
Device IDs: ['device-1', 'shellypro3em-ec4609c439c2', 'shellypro3em-ec4609c439c2']
Skip Test: False
Testing powermeter configuration...
Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): -47.0W
Starting device: ct001
CT001 Settings for device-1:
Disable Sum Phases: False
Disable Absolute Values: True
Poll Interval: 1
Starting device: shellypro3em_old
UDP server is listening...
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c2
TCP server is listening...Shelly emulator listening on UDP port 1010...
Starting device: shellypro3em_new
Shelly Pro 3EM Settings:
Device ID: shellypro3em-ec4609c439c2
Shelly emulator listening on UDP port 2220...

tomquist, 2025-06-07:
Which device are you using and what firmware version do you have? Also please check the [FAQ](https://github.com/tomquist/b2500-meter#frequently-asked-questions-faq). It provides answers to some of the common issues.

univers718, 2025-06-07:
Marstek Venus 5.12 kWh.
Firmware version V151

closed 2026-05-16 by tomquist