tomquist/AstraMeter #133: Fail to read B-Control Modbus Power Meter EM300

TheItschi, 2025-06-21:
When trying to use my B-Control Modbus Power Meter EM300 (configured as Modbus over TCP) I get the following erro: 

DEBUG:b2500-meter:Error: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'unit'

Using my Tasmota Power Meter works, but I trust my EM300 more

I'm using the Direct Installation on a Windows 11 system with Python 3.10.4. I needed to trick this because my current Phyton on this machine is 3.8.10. Unfortunately the scripts does not run with this version and gives the following error:

Traceback (most recent call last):
  File "C:\b2500-meter-main\main.py", line 9, in <module>
    from shelly import Shelly
  File "C:\b2500-meter-main\shelly\__init__.py", line 1, in <module>
    from .shelly import Shelly
  File "C:\b2500-meter-main\shelly\shelly.py", line 9, in <module>
    class Shelly:
  File "C:\b2500-meter-main\shelly\shelly.py", line 11, in Shelly
    self, powermeters: list[(Powermeter, ClientFilter)], udp_port: int, device_id
TypeError: 'type' object is not subscriptable

I think it hast to do with a chnge in List handling introduced with 3.9 onward. I'm not deep into Phyton to fix this in the code, maybe someone else can giv a hint.

The configuration is:

[MODBUS]
HOST = 192.168.100.119
PORT = 502
#UNIT_ID = 1
#ADDRESS = 0
#COUNT = 1
# Per-powermeter throttling override (optional)
#THROTTLE_INTERVAL = 1

No idea what the other parameters are. There is no info about those for my EM300.

Regrads, Itschi

TheItschi, 2025-06-22:
If it helps, I could offer to make my EM300 accessible via internet to debug the problem.

tomquist, 2025-06-22:
I found some documentation for this device here:
https://soldraft.com/spx_web_files/shop/file/6/Montageanweisung%20B-Control%20EM300LR.pdf

It looks like the value you need spans two registers, which is currently not supported. I am updating the modbus integration here: https://github.com/tomquist/b2500-meter/pull/134

Once that’s merged and a new version is released you should be able to use this config:
```
[MODBUS]
HOST = 192.168.100.119
PORT = 502
UNIT_ID = 1                  # Usually 1 unless explicitly changed
ADDRESS = 9                  # Register for OBIS 1.7.0 (Wirkleistung Total)
COUNT = 2                    # 2 registers for 32-bit float
DATA_TYPE = FLOAT32          # Value is a 32-bit float
BYTE_ORDER = BIG             # Each 16-bit register is big-endian
WORD_ORDER = BIG             # High register first (standard for this device)
```

TheItschi, 2025-06-22:
Just found that there is a JSON http interface as well. See the example Python example blow:

`#
# Copyright: Kilian Knoll, 15.12.2018
# Utility to parse EnergyManager EM300LR using the JSON API
#
#
#  
#  This program is free software: you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation, either version 3 of the License, or
#  (at your option) any later version.
#
#  This program is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#  GNU General Public License for more details.
#
#  You should have received a copy of the GNU General Public License
#  along with this program.  If not, see <http://www.gnu.org/licenses/>.
#
#
#  Please note that any incorrect or careless usage of this module as well as errors in the implementation can damage your Energy Manager!
#  Therefore, the author does not provide any guarantee or warranty concerning to correctness, functionality or performance and does not accept any liability for damage caused by this module, examples or mentioned information.
#  Thus, use it at your own risk!
#
#
#  Purpose: 
#  Returns:
#     Returnvalue: 0    (Everything all right)
#     Returnvalue: -1   (Crap happened)
#     em300data:        (Empty list in case Returnvalue =-1)
#     em300data:        (Full list of key-value pairs in case Returnvalue = 0)
#
# Tested with:
#           python 3.5, 3,7
#           B-control Energy Manager EM300, Firmware Version 2.04
# Based on: https://www.tq-automation.com/Service-Support/Downloads/Downloads-Energiemanagement
# Using the JSON Documentation: https://www.tq-automation.com/content/download/10996/file/TQ%20Energy%20Manager%20-%20JSON-API.0104.pdf
#

# Please change the following values to reflect your environment:
Em_IP_Adress = "192.168.178.40"                     #IP-Adress of the Energy Manager
EM_Serialumber= "12345678"							#Serial Number can be found accessing the Energy Manager´s Web Page under  Settings - Serial number
EM_Password= "YourPassword"						    #This is the password you have specified on the Energy Manager´s Web Page (without quotes)
# End Configurable Parameters 
#
import requests
import time 
import json
import time
import datetime
import logging
from pprint import pprint

myparams= {'login': EM_Serialumber, 'password': EM_Password, 'save_login': '1'}

headers = {
    "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64; "
}
mySession = requests.Session()

def changeobiskeys(myjsoninput):
    myjsoninput["Active power_incoming"] =  myjsoninput.pop("1-0:1.4.0*255")
    myjsoninput["Active energy_incoming"] =  myjsoninput.pop("1-0:1.8.0*255")
    myjsoninput["Apparent power-"] =  myjsoninput.pop("1-0:10.4.0*255")
    myjsoninput["Apparent energy-"] =  myjsoninput.pop("1-0:10.8.0*255")
    myjsoninput["Power factor"] =  myjsoninput.pop("1-0:13.4.0*255")
    myjsoninput["Supply frequency"] =  myjsoninput.pop("1-0:14.4.0*255")
    myjsoninput["Active power- "] =  myjsoninput.pop("1-0:2.4.0*255")
[... 178 lines cut ...]
 'Reactive energy- L2': 11421844.1,
 'Reactive energy- L3': 11545052.6,
 'Reactive energy_incoming': 2153222.7,
 'Reactive energy_incoming L1': 146009.3,
 'Reactive energy_incoming L2': 1588442.9,
 'Reactive energy_incoming L3': 1708789.9,
 'Reactive power-': 2477.5,
 'Reactive power- L1': 1060.4,
 'Reactive power- L2': 792.5,
 'Reactive power- L3': 624.6,
 'Reactive power_incoming': 0,
 'Reactive power_incoming L1': 0,
 'Reactive power_incoming L2': 0,
 'Reactive power_incoming L3': 0,
 'Supply frequency': 49.996,
 'Voltage L1': 233.697,
 'Voltage L2': 230.14,
 'Voltage L3': 232.636,
 'serial': '70808598',
 'status': 0}
----------------End - Values from Energy Manager ----------------
Two Specific values from array....
Energy to Grid   (Obis code: 1-0:2.8.0) 56308799.8
Energy from Grid (Obis code: 1-0:1.8.0) 42735020.5
`

TheItschi, 2025-06-25:
Just added the implementation for the EM300 Energy Meter interface:

---
from .base import Powermeter
import requests
import time
import json

class Em300(Powermeter):
    def __init__(self, ip: str, json_power_calculate: bool):
        self.ip = ip
        self.json_power_calculate = json_power_calculate
        self.session = requests.Session()

    def get_json(self, path):
        import requests
        import json

        headers = {
            "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64; "
        }

        mySession = requests.Session()
        r = mySession.get('http://'+ self.ip + '/start.php',  headers=headers)
        r = mySession.get('http://'+ self.ip +'/mum-webservice/data.php',  headers=headers)
        em300data = json.loads(r.text)

        if (len(em300data) == 0):
            em300data={}
            logger.debug(f"No data from EM300 received")

        return (em300data)    

    def get_powermeter_watts(self):
        response = self.get_json(self)
        if not self.json_power_calculate:
            return [int(response["1-0:1.4.0*255"])]
        else:
            power_in = response["1-0:1.4.0*255"]
            power_out = response["1-0:2.4.0*255"]
            return [int(power_in) - int(power_out)]
--- 

How can I contribute to your code?

tomquist, 2025-06-25:
With the upcoming version you can now also use a generic http json Powermeter which would look something like this for your EM300:
```
[JSON_HTTP]
URL = http://192.168.1.100/mum-webservice/data.php
JSON_PATHS = $.["1-0:1.4.0*255"]
```

TheItschi, 2025-06-25:
Sounds great, but keep in mind that for the EM300 it needs two subsequent requests (or even three if you use a password) to different URLs to get the data. So a generic approach might not be the best option for this power meter.