tomquist/AstraMeter #141: B2500-meter mit zwei Speichern

Uenne, 2025-06-24:
## What's the issue?
Ich habe an meinem BKW zwei B2500 unabhängig installiert. An einem habe ich das Script ans Laufen gebracht, mit ShellyPro 3EM. Als Sollvorgabe nutze ich eine gleitenden Mittelwert (aus Hitachi-Lesekopf), der alle 5 s aktualisiert wird. Der Wert wird halbiert, da ich zwei Speicher ansteuern will und diese sich nicht hochschaukeln sollen. Zielwert ist also der halbierte gleitende Mittelwert.
Wie kann ich den zweiten Speicher verbinden? Meine Idee ist: 2. Zielordner erstellen und einen anderen Shelly (z.B. ShellyEMgen3) zu simulieren.
Geht das so?

zweite Frage:
Gibt es eine Möglichkeit, die Werte des B2500 in IObroker zu bringen?
Anfangs habe ich einen ESP (ebenfalls das Projekt von Tomquist) zum Auslesen genutzt, aber diesen jetzt abgeklemmt, damit keine Störung mit dem B2500-meter auftritt.

..und noch eine Frage:
Das Projekt hm2mqtt könnte meine zweite Frage lösen? Ich verstehe das Projekt so, das Rechner und B2500 mqtt-mäßig über BLE kommunizieren? Was ist, wenn Rechner und B2500 zu weit auseinander stehen? 

## Your setup
- IObroker
- B2500
- Hitachi-Lesekopf

thowiegit, 2025-06-26:
Mehrere Meter machen nur dann Sinn, wenn Du den Speichern Messwerte aus unterschiedlichen Quellen senden möchtest.
B2500-meter kann aber auch einfach so - mit nur einem konfigurierten Meter - von mehr als einem Speicher genutzt werden.
Einfach auf beliebig vielen Speichern den Selbstanpassungsmodus aktivieren und immer den gleichen SmartMeter Typ auswählen.

Bei mir laufen aktuell 4 der Speicher genau so.

tomquist, 2025-06-27:
@Uenne Du kannst auch mehrere Shellys aus unterschiedlichen Quellen mit einer Instanz von B2500-Meter emulieren. Wie das geht ist hier dokumentiert: https://github.com/tomquist/b2500-meter/tree/1.0.7?tab=readme-ov-file#multiple-powermeters

Allerdings bin ich mir nicht sicher ob ein simples halbieren der Werte ausreicht um ein Aufschaukeln zu verhindern. Besser wäre es kontinuierlich die Einspeisung des Speichers mit dem Verbrauch zu verrechnen und so dann damit den Soll-Wert mit halber Aufteilung zu berechnen. Ist vermutlich aber etwas komplizierter.

@thowiegit Funktioniert das bei dir wirklich gut mit mehreren Speichern aus der gleichen Quelle? Wie verteilt sich dabei normalerweise die Last?

thowiegit, 2025-06-28:
@tomquist ich habe beides versucht.
Letztlich ist es egal, ob man jedem Speicher nur die halbe oder allen Speichern die volle Last sendet.
Es ist auch egal ob man nur einen Shelly simuliert, oder jedem Speicher seinen eigenen Shelly.
So lange alle ihre Daten aus der gleichen Quelle beziehen, also auch alle Speicher in den gleichen Stromkreis einspeisen ist nicht vorhersagbar welcher Speicher wie viel der Last tragen wird. Das ist völlig erratisch.
Eine halbwegs beständige Gleichverteilung ist nur bei entsprechend hohen Lasten zu beobachten.
Ich habe das setup daher wieder auf das simpelste zurückgefahren.
Alle Speicher bekommen den gleichen simulierten Shelly zu sehen und reagieren darauf wie sie grad mögen.
Am Ende hängt es vom Timing ab, wer eine zehntel Sekunde eher den aktuellen Wert liest wird auch mehr Last tragen, bis die Last wieder sinkt, dann ist der entsprechende Speicher auch der erste, der seine Ausgabe reduziert.

tomquist, 2025-06-28:
Ja. Das funktioniert eben nur für DC Speicher wie der B2500, die ihren Strom nicht aus dem AC Netz beziehen. Ansonsten würde es unter Umständen nämlich dazu kommen, dass ein Speicher aus dem anderen lädt.

Um eine korrekte Verteilung der Last umzusetzen, gibt es im Prinzip 2 Möglichkeiten:
- Jeder Speicher bekommt seine eigene(n) Phase(n) zugewiesen. Dadurch haben die Aktionen eines Speichers keinen Einfluss auf die anderen Speicher. Das ist eine schon möglich, wenn man einen Sensor hat, der die einzelnen Phasen auslesen kann.
- Es wird jedem Speicher ein virtueller Sensor gegeben, der die Einspeisung der anderen Speicher mit einrechnet.

Sobald die CT002 Emulation funktioniert, gibt es noch eine weitere Möglichkeit, denn darüber kann jedem Speicher einzeln gesagt werden, wie viel er einspeisen soll, wodurch man die Möglichkeit hat, den Verbrauch mit der aktuellen Einspeisung zu verrechnen um eine genaue Aufteilung zu realisieren.

Uenne, 2025-07-01:
Danke für die verschiedenen Kommentare. Das Verbinden beider B2500 mit einem Shelly hat geklappt.
Ich sehe allerdings ein deutliches Schwingen, auch bei geglätteten Zahlen. Ich kann zwar die Zeit einer Schwingung beeinflussen, aber nicht die Amplitude. Da arbeite ich noch dran. Jetzt aber erstmal Urlaub, danach geht es weiter.

tomquist, 2025-07-01:
Der Parameter „Throttle Interval“ hilft im das Schwingen zu verhindern.