tomquist/AstraMeter #154: B2500 nur volle Leistung oder minimum

DustinThal, 2025-06-28:
### Question

Hi, 
danke erstmal für dieses tolle add-on.
Ich habe gestern mein B2500 bekommen und auch direkt einrichten wollen.
Da schon 2 kWh im Speicher waren, konnte ich mich direkt austoben. 
nur leider scheint irgendwas überhaupt nicht zu stimmen. 
ich bin nun auf der 110.9. und entweder er speist garnicht ein, mit 80 watt oder mit 800. dazwischen nur selten.

[image]

obwohl diese werte gegeben werden:

[image]

Ist das normal? Was habe ich vielleicht übersehen?

Vielen Dank schon einmal für die Hilfe

### Device Model (optional)

B2500 HMJ-2

### Firmware Version (optional)

110.9

tomquist, 2025-06-28:
Wie sieht denn die Konfiguration aus?

DustinThal, 2025-06-28:
Hi, 
leider muss ich gleich los, aber schön das man so schnell eine Rückantwort erhält, hier die Konfiguration:

[image]

[image]

DustinThal, 2025-06-28:
Und hier noch vom add-on
[image][image]

Netzfluss ist ein Wert der sich aus Verbrauch (Summe vieler Shelly's) und B2500 Output errechnet, und negativ sein kann wenn ich mehr vom B2500 bekomme Anka ich brauche.

DustinThal, 2025-06-28:
Da ich erst ab ca 13 Uhr wirklich Leistung habe werdenden berichten, stelle nun auf 10. Danke für den Hinweis.

DustinThal, 2025-06-28:
Statt den Output zu nehmen wäre also ein Zwischenstecker sinnvoll der das messen kann .... da die schneller zurückmelden, hab ich's richtig verstanden ?

DustinThal, 2025-06-28:
Leider aktuell nicht möglich, wir arbeiten aber daran. 
Hat dies irgendeine Relevanz wenn ich mit Shelly arbeite? Finde es in der App leider nicht zum vergleichen:
[image]

DustinThal, 2025-06-29:
Leider war es nicht erfolgreich mit dem heraufsetzen des Throttling Intervals auf 10 Sekunden, ich hab es mal auf 30 gesetzt, jetzt hab ich keine 800 Watt peaks mehr aber die 80 Watt downs ... bzw, gerade seit 10 Minuten nicht mehr ....

[image]
Blau ist Ausgang B2500, rot Eingang EZ1, grün mein aktueller (noch berechneter) Verbrauch.
Shelly kommt am Dienstag rein.

DustinThal, 2025-06-29:
Seit 17:42 ist er Akku voll, könnte damit zusammenhängen

tomquist, 2025-06-29:
Gibst du den Wert von Grün an B2500-Meter? Seltsam ist, dass eine Veränderung der Ausgangsleistung am B2500 keine Auswirkung auf die Grüne Linie hat. Das sollte aber so sein, wenn du Grün an B2500-Meter gibst.

DustinThal, 2025-06-29:
Ganz am Anfang ja, brachte aber nicht den erwünschten Erfolg.
Danach habe ich über grün und die Solarleistung des EZ1 mir einen Wert für Netzfluss erstellt, dieser kann nun auch negativ werden, wenn mehr vom EZ1 ins netz geht als ich verbrauche. diesen gebe ich über das b2500-meter an den B2500.

tomquist, 2025-06-29:
Mir ist immernoch nicht ganz klar welchen Wert du jetzt an B2500-Meter gibst. Kannst du mal die Berechnung hier posten? Ich vermute nach wie vor, dass da etwas nicht stimmt.

tomquist, 2025-06-29:
Nutzt du hm2mqtt oder wie holst du die Daten von B2500? Wie oft fragst du die Daten ab?

Der Grund warum ich frage ist, dass die Blaue Kurve so aussieht als würden die Daten nur ein Mal pro Minute aktualisiert werden.

tomquist, 2025-06-29:
Summe aller Verbraucher minus Einspeisung des B2500 wie du es oben in dem Template hast ist schon richtig, aber eine Aktualisierungsrate von 60s würde das Problem sehr gut erklären. Der Speicher ruft die Daten, wenn er nicht gethrottled wird, jede Sekunde ab. Er sieht dann, dass der Verbrauch >0 ist und erhöht dann die Ausgangsleistung. Beim nächsten Abrufen des Verbrauchs sieht er dann immernoch den gleichen Wert >0 und erhöht weiter, so lange bis dann hm2mqtt nach 60s die Daten aktualisiert, woraufhin dein errechneter Wert negativ wird, woraufhin der Speicher dann anfängt die Ausgangsleistung zu reduzieren, bis auf das Minimum von 80W. Nach 60s aktualisiert dann hm2mqtt wieder und dein errechneter Wert ist wieder >0 und das ganze Spiel beginnt wieder von vorne.

DustinThal, 2025-06-29:
Das war es, hab am falschen Ende gesucht, vielen vielen Dank.
Sieht nun viel besser bis perfekt aus 🫣

[image]

tomquist, 2025-06-29:
Wenn du das Aktualisierungsinterval von hm2mqtt auf 5s hast, dann solltest du das throttle interval mindestens auch auf 5s stellen, sonst kommt es wieder zu einem oversteering

DustinThal, 2025-06-29:
jap, war vllt nicht der beste moment dass der Speicher ausgestiegen ist, kann auch keinen Wechselrichter mehr konfigurieren ... doof, dass die Setting in der App vom Server kommen ..... 
danke erstmal, ich bin für heute raus.

DustinThal, 2025-07-06:
So, hab nen Shelly, direkt an die Batterie, funktioniert klasse. Leider muss ich nun erstmal Kalibrieren ....[image]

closed 2025-07-06 by DustinThal