tomquist/AstraMeter #156: eu.hamedata.com down -> keine Regelung mehr

iopodx, 2025-06-29:
Hallo,

falls es noch jemand betroffen hat: die eu.hamedata.com ist gerade "down".

Ich habe das mal fix analysiert und gesehen, dass die B-2500 mit 110 Aufrufe an

http://eu.hamedata.com/app/neng/getDateInfoeu.php?uid=[redacted]&fcv=[redacted]&aid=HMJ-2&sv=110&sbv=9&mv=105  senden.

Ich habe das in meinem DNS eben auf us.hamedata.com umgebogen (3.133.77.112), nun funktioniert der Spaß wieder. Die response ist 

> _2025_06_29_23_55_17_04_0_0_0

Ich wollte das nur kurz dokumentieren, eigentlich gehört es nicht so richtig zum Projekt, aber vielleicht wundert sich ja der eine oder andere warum die "lokale" Regelung nicht funktioniert. Ich denke ich werde die Tage mal eine lokale Lösung für diese simplen "NTP"-Anfragen implementieren.

muhkuhgh, 2025-06-30:
Jetzt bin ich ehrlich gesagt nur noch verwirrter als vorher.. mein Shelly-Emulator sendet ja gar keine Zeit mit.. sollte also problemlos davon unabhängig funktionieren.. (was er aber ja faktisch nicht hat).. wohingegen ich problemlos Zeitprofile anlegen konnte und diese dann auch zur korrekten Uhrzeit geschaltet haben. Völlige Marstek Madness 🤯

iopodx, 2025-06-30:
Wir können nur spekulieren was diese Anfrage soll oder warum sie überhaupt so oft gesendet wird. Vielleicht ist es gewollt um Marstek "Kontrolle" zu geben oder es ist einfach nur schlecht implementiert und es gibt kein vernünftiges Error-Handling dafür.

Mir wäre es ohnehin am Liebsten den Geräten perspektivisch den Zugriff zum Internet zu nehmen, wenn Zeit ist werde ich daher mal schauen was noch so für Daten rausgehen und wie man das lokal lösen kann.

michaeljk, 2025-06-30:
Waren auch Besitzer eines physikalisch vorhandenen Shelly von dem Ausfall betroffen? Das würde ja dann bedeuten, das bei Ausfall der Cloud-Server die Nulleinspeisung insgesamt nicht mehr funktioniert. Und viel schlimmer, das diese ohne Internet sowieso nicht läuft.