tomquist/AstraMeter #167: B2500-Meter with 2x B2500D - uneqal amounts of power

fabianwimberger, 2025-07-11:
**What's the issue?**  
I am experiencing issues connecting two B2500D units to a single `b2500-meter` instance. When only one instance of `b2500-meter` is run, only one of the B2500D units connects.

My current workaround involves running **two separate `b2500-meter` instances**, each connected to a different Shelly device (Shelly Pro 3EM and Shelly EM Gen3), both listening to the **same smart meter MQTT topic**. While this allows both units to connect, they often deliver **significantly different amounts of power to the grid**, despite both being set to **"Auto" mode**.

The smart meter provides updates every 5 seconds, and the `throttle_rate` is set accordingly.

---

**Your setup**  
- **Installation method:** 2× `b2500-meter` (Docker)  
- **Storage system:** 2× B2500D (Firmware Version 110)  
- **Power meter source:** Smart meter (Austria) publishing MQTT data every 5 seconds  
- **Current workaround:**  
  - `b2500-meter-1` connected to **Shelly Pro 3EM**  
  - `b2500-meter-2` connected to **Shelly EM Gen3**  
  - Both listen to `sma/net_power` MQTT topic

---

**Configuration**

<details>
<summary>Config 1: Shelly EM Gen3</summary>

```ini
[GENERAL]
DEVICE_TYPE = shellyemg3
SKIP_POWERMETER_TEST = False
THROTTLE_INTERVAL = 5

[MQTT]
BROKER = 192.168.178.99
PORT = 1883
TOPIC = sma/net_power
USERNAME = XXXX
PASSWORD = XXXX
```

</details>

<details>
<summary>Config 2: Shelly Pro 3EM</summary>

```ini
[GENERAL]
DEVICE_TYPE = shellypro3em
SKIP_POWERMETER_TEST = False
THROTTLE_INTERVAL = 5

[MQTT]
BROKER = 192.168.178.99
PORT = 1883
TOPIC = sma/net_power
USERNAME = XXXX
PASSWORD = XXXX
```

</details>

---

**Logs from b2500-meter**

```log
Throttling: Waiting 5.0s before fetching fresh values...
Throttling: Fetched fresh values after 0.0s interval: [-5.0]
Throttling: Waiting 5.0s before fetching fresh values...
Throttling: Fetched fresh values after 0.0s interval: [-6.0]
Throttling: Waiting 5.0s before fetching fresh values...
Throttling: Fetched fresh values after 0.0s interval: [0.0]
...
Throttling: Fetched fresh values after 0.0s interval: [13.0]
Throttling: Waiting 5.0s before fetching fresh values...
```

**Logs from b2500-meter-1**

```log
Throttling: Fetched fresh values after 0.0s interval: [-2.0]
Throttling: Waiting 5.0s before fetching fresh values...
Throttling: Fetched fresh values after 0.0s interval: [0.0]
...
Throttling: Fetched fresh values after 0.0s interval: [-11.0]
Throttling: Waiting 5.0s before fetching fresh values...
```

---

**Output of esphome-b2500**

<details>
<summary>Device Overview</summary>

```
Firmware Version: 2025.6.3 (Jul 8 2025, 10:31:53)
Device Type: ESP32-S3-DevKitC-1 by Espressif
MAC Address: [REDACTED]
MQTT Connected: Yes
WiFi SSID: [REDACTED]
```

</details>

<details>
<summary>Devices</summary>

```
🔋 B2500-1
- Battery Level: 65.0%
- Battery Capacity: 1456 Wh
- BLE Connected: ✅
- Charge Mode: SimultaneousChargeAndDischarge
- Power In: 48 W
- Power Out: 114 W
- Temperature 1 / 2: 20.0°C / 21.0°C
- Total Daily Energy In / Out: 1.84 kWh / 0.59 kWh
- SmartMeter Power Out: 114 W

🔋 B2500-2
- Battery Level: 80.0%
- Battery Capacity: 1792 Wh
- BLE Connected: ✅
- Charge Mode: SimultaneousChargeAndDischarge
- Power In: 49 W
- Power Out: 162 W
- Temperature 1 / 2: 21.0°C / 22.0°C
- Total Daily Energy In / Out: 1.93 kWh / 0.39 kWh
- SmartMeter Power Out: 162 W
```

</details>

<details>
<summary>Configuration</summary>

```
Depth of Discharge: 80
Timers:
- Timer 1: 400 W
- Timers 2–5: 80 W
```

</details>

<details>
<summary>Diagnostics</summary>

```
🔧 B2500-1
- Device ID: [REDACTED]
- Firmware: 110 / FC41D-Firmware 202310231502
- Last Response: 2025-07-11T14:13:24
- Device Type: HMJ-2
- Region: EU

🔧 B2500-2
- Device ID: [REDACTED]
- Firmware: 110 / FC41D-Firmware 202310231502
- Last Response: 2025-07-11T14:13:23
- Device Type: HMJ-2
- Region: EU
```

</details>

fabianwimberger, 2025-07-12:
@droidbanana that's interesting, do you also use a relatively high throttling number (5s)?

fabianwimberger, 2025-07-13:
Alright, thank you for your insight! I will close this issue as duplicate - the other topic contains much more info and refers to the same issue. Thank you anyways!

closed 2025-07-13 by fabianwimberger

tomquist, 2025-07-19:
@fabianwimberger The new version 1.0.8 contains improvements to handling multiple devices when using throttle interval. With that version it should not longer be necessary to use multiple instances of B2500-Meter.

fabianwimberger, 2025-07-19:
@tomquist thanks for the great work! If you need somebody to test, I will gladly do it. In the meantime, i improved my dual-b2500 setup to actively handle power sharing between devices to keep their battery balanced.

Short summary of script:
- **Real-time balancing**: Distributes power based on battery levels (higher battery = more load)
- **Failover support**: Routes all power to remaining storage if one goes offline (via smartmeter/connected status)
- **Hardware limits**: 40W-400W per storage (requests 500W in MAX mode for guaranteed full unlock)
- **MQTT-based**: Reads storage status, publishes power adjustments to mqtt where b2500-meter can transmit to specific storages.

Would you be interested in a PR to integrate this power balancing functionality, or would you prefer to keep the scope focused on single-device management?