tomquist/AstraMeter #169: Works for 1 houre

ChristophHerrx, 2025-07-14:
## What's the issue?
The B2500 finds the Shelly 3em pro and it works fine but after ~one hour the battery changed back to constant power. I can't reach the battery by the app anymore.

## Your setup
- **Installation method**:  Home Assistant Add-on/Proxmox
- **Storage system**:   B2500, v113
- **Power meter source**: Tasmota with Hitchi IR 

## Configuration
<html>
<body>

update.marstekMarstek | off | auto_update: false display_precision: 0 installed_version: 1.0.7 in_progress: false latest_version: 1.0.7 release_summary: null release_url: null skipped_version: null title: B2500 Meter update_percentage: null friendly_name: Marstek supported_features: 25
-- | -- | --

</body>
</html>

Optionen
Power Input Alias
sensor.tasmota_sm_56_7_0

Power Output Alias

Poll Interval
1

Allow Negative Values = Off

Emulated Devices
shellypro3em

Throttling Interval
0

Log Level

info

## Error messages or logs
```
-----------------------------------------------------------
 Add-on: B2500 Meter
 "Replicates a Smart Meter device for a B2500 energy storage system while allowing integration with various smart meters."
-----------------------------------------------------------
 Add-on version: 1.0.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 16.0  (amd64 / qemux86-64)
 Home Assistant Core: 2025.7.1
 Home Assistant Supervisor: 2025.07.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
Log level is set to INFO
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[GENERAL]
DEVICE_TYPE=shellypro3em
POLL_INTERVAL=1
DISABLE_ABSOLUTE_VALUES=false
THROTTLE_INTERVAL=0
ENABLE_HEALTH_CHECK=true
[HOMEASSISTANT]
IP=supervisor
PORT=80
API_PATH_PREFIX=/core
ACCESSTOKEN=3833d037e56cde0338b1c59ebd0ccd12cc61647cb454423c267debea7833400cb9504cbe9b5b0be344d7905ce8bcf475e10953cf40e7b584
POWER_CALCULATE=False
CURRENT_POWER_ENTITY=sensor.tasmota_sm_56_7_0
[22:45:35] INFO: Waiting for Home Assistant to be ready...
[22:45:35] INFO: Home Assistant is ready! Proceeding with B2500 Meter startup...
[22:45:35] INFO: Starting B2500 Meter with log level: info
INFO:b2500-meter:startet b2500-meter application
INFO:b2500-meter:Device Types: ['shellypro3em_old', 'shellypro3em_new']
INFO:b2500-meter:Device IDs: ['shellypro3em-ec4609c439c1', 'shellypro3em-ec4609c439c1']
INFO:b2500-meter:Skip Test: False
INFO:b2500-meter:Starting health check service...
INFO:b2500-meter:Health check service started on 0.0.0.0:52500
INFO:b2500-meter:Health check service started successfully
INFO:b2500-meter:Shelly emulator listening on UDP port 1010...
INFO:b2500-meter:Shelly emulator listening on UDP port 2220...
INFO:b2500-meter:Health check service started successfully
INFO:b2500-meter:Shelly emulator listening on UDP port 1010...
INFO:b2500-meter:Shelly emulator listening on UDP port 2220...

```

## Additional info

[update.marstek](http://192.168.178.74:8123/developer-tools/state#)
Marstek
off	auto_update: false
display_precision: 0
installed_version: 1.0.7
in_progress: false
latest_version: 1.0.7
release_summary: null
release_url: null
skipped_version: null
title: B2500 Meter
update_percentage: null
friendly_name: Marstek
supported_features: 25

ChristophHerrx, 2025-07-23:
Yes I think it's not a Shelly issue. I'll wait for an Marstek update..

BaffoBill, 2025-08-05:
I experience this issue about once a week. Starting at 00:00  CEST (device time: 22:00 ), the B2500 no longer transmits data via Wi-Fi or Bluetooth.
Restarting the B2500 resolves the issue.
My firmware: V113.1
I hope for a better FW from Marstek...