tomquist/AstraMeter #171: Marstek Jupiter speist nichts ein

mcc1710, 2025-07-16:
## What's the issue?
Ich berechne den Gesamtverbrauch im Haus und habe hierfür einen Sensor angelegt. Dieser kann sowohl positive als auch negative Werte ausgeben.

Der emulierte Shelly wird vom Marstek Jupiter erkannt allerdings reagiert dieser nicht darauf. Sprich es wird nichts eingespeist. 
Welche Einstellung soll ich anpassen? 
Danke für die Unterstützung.

## Your setup
- Marstek Jupiter
- Errechneter Gesamtverbrauch

## Configuration
- Poll Intervall: 1
- Throttling Intervall: 0
- Emulated Device: shellypro3em
- Marstek SW Version: V134

## Additional info

[image]
[image]
[image]

[logs.txt](https://github.com/user-attachments/files/21246969/logs.txt)

mcc1710, 2025-07-16:
Hi
Danke für den Hinweis. Ich werde mir die Verzögerung anschauen und mich nochmal melden. 
Aktuell berechne ich den Gesamtverbrauch so: 

[image]

tomquist, 2025-07-16:
@schumischumi Deinem sensor fehlt noch den Wert für die aktuelle Einspeisung des Wechselrichters abzuziehen. So kann der Speicher nicht regeln. Der Speicher versucht kontinuierlich auf 0W zu regeln, wenn sich dein errechneter Wert aber mit einer Erhöhung der Ausspeiseleistung nicht verringert, dann erhöht der Speicher die Ausgabe so lange bis er das Maximum erreicht hat.

schumischumi, 2025-07-16:
Ich habs jetzt über den neuen sensor gemacht und er pendelt sich langsam ein. sieht gut aus und danke dir!

mcc1710, 2025-08-20:
Hi 
Ich habe gerade das neuste Update installiert und alles funktioniert wie es soll. 
Vielen Dank für die Unterstützung.

closed 2025-08-20 by mcc1710