tomquist/AstraMeter #184: b2500-meter does not find real shelly3EM

LeoSum8, 2025-08-12:
## What's the issue?

## Your setup
- **Installation method**: Docker
- **Storage system**: B2500 (firmware version not yet known)
- **Power meter source**: Shelly 3EM

## Configuration
```ini
[GENERAL]
DEVICE_TYPE = shellypro3em
SKIP_POWERMETER_TEST = False
DISABLE_SUM_PHASES = False
DISABLE_ABSOLUTE_VALUES = False
POLL_INTERVAL = 1
THROTTLE_INTERVAL = 0

[SHELLY]
TYPE = 3EM
IP = 192.168.217.49
USER = admin
PASS = admin
METER_INDEX = meter1
```
initially I didn't have user and password set in the shelly, but setting it didn't make it work either.

b2500-meter container running via:
```
docker run --name b2500meter --network host -v /var/b2500-meter_data/config.ini:/app/config.ini ghcr.io/tomquist/b2500-meter:latest
```

## Error messages or logs
```
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:startet b2500-meter application
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:Device Types: ['shellypro3em_old', 'shellypro3em_new']
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:Device IDs: ['shellypro3em-ec4609c439c1', 'shellypro3em-ec4609c439c1']
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:Skip Test: False
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:Starting health check service...
Aug 12 14:02:06 coreos.lan docker[1371774]: INFO:b2500-meter:Health check service started on 0.0.0.0:52500
Aug 12 14:02:07 coreos.lan docker[1371774]: INFO:b2500-meter:Health check service started successfully
Aug 12 14:02:07 coreos.lan docker[1371774]: INFO:b2500-meter:Retrying powermeter test in 5 seconds...
Aug 12 14:02:12 coreos.lan docker[1371774]: INFO:b2500-meter:Retrying powermeter test in 5 seconds...
Aug 12 14:02:17 coreos.lan docker[1371774]: INFO:b2500-meter:Retrying powermeter test in 5 seconds...
Aug 12 14:02:22 coreos.lan docker[1371774]: ERROR:b2500-meter:Failed to test powermeter after 4 attempts: Expecting value: line 1 column 1 (char 0)
Aug 12 14:02:22 coreos.lan docker[1371774]: INFO:b2500-meter:Health check service stopped
Aug 12 14:02:23 coreos.lan systemd[1]: docker.b2500meter.service: Main process exited, code=exited, status=1/FAILURE

```

## Additional info
I can ping my shelly from the docker host just fine.
How can I further pinpoint why b2500-meter doesn't connect to my shelly?

tomquist, 2025-08-12:
`METER_INDEX` must either be a number representing the phase you want to use or be removed in case you want to use all phases.

LeoSum8, 2025-08-12:
Ah, that did it. I must have misread the docs. Sorry for stealing your time and thanks a lot!

closed 2025-08-12 by LeoSum8