tomquist/AstraMeter #200: No data in Marstek App since V139 on Marstek Venus E V3

deepblue120, 2025-09-17:
## What's the issue?
Daignose in Masrtek App lauft auf Fehler Emuliertes Gerät online, aber meldet keine Daten

## Your setup
Home Assistant Add-on 
Venus E V3 FW V139 -->
- **Power meter source**:  Home Assistant sensor

## Configuration
power_input_alias: sensor.alpha_ess_netzbezug_leistung_vom_netz
power_output_alias: sensor.alpha_ess_netzbezug_leistung_zum_netz_positiv
poll_interval: 1
disable_absolute_values: true
device_types: shellyproem50
throttle_interval: 1.5
log_level: info
shelly_push_mode: true
SHELLY_PUSH_TARGET_IP: 192.168.178.251
SHELLY_PUSH_TARGET_PORT: 22222
SHELLY_PUSH_INTERVAL: 1

```

## Error messages or logs

```

## Additional info
Seems to happen since update to V139 on Marstek and LAN support

deepblue120, 2025-09-18:
Maybe some more informations. With update to firmware V139 the LAN port was activated. My Marstek is now connected via LAN and WLAN in the same network with two IPs. B2500 fails to connect in the Marstek app. My Shelly 3PM Pro hard-wired device works fine in the Marstek app.
When disconnecting the LAN port the emulated 3EM is working again and the battery is receiving correct data.

So it seems that the LAN port is the problem.

Scaro657, 2025-11-21:
It appears that when connected to LAN, the JSON datagram from the Shelly is expected to be at least 250 characters long. I used the JSON object from this project for inspiration, and this works fine when Venus is connected via Wifi, but stopped working when connected to LAN.

sln302, 2025-12-04:
have my marstek v139 connected through wifi. Trying to get a shellypro3em emulation working, but it's not detected by the marstek app it seems... I do run in pipenv.

sln302, 2025-12-05:
I enabled logging:

INFO:b2500-meter:startet b2500-meter application
INFO:b2500-meter:Device Types: ['shellypro3em_old', 'shellypro3em_new']
INFO:b2500-meter:Device IDs: ['shellypro3em-ec4609c439c1', 'shellypro3em-ec4609c439c1']
INFO:b2500-meter:Skip Test: False
INFO:b2500-meter:Starting health check service...
INFO:b2500-meter:Health check service started on 0.0.0.0:52500
DEBUG:b2500-meter:Health check request received from 127.0.0.1:52838
DEBUG:b2500-meter:Health check endpoint test passed
INFO:b2500-meter:Health check service started successfully
DEBUG:b2500-meter:Testing powermeter configuration... (attempt 1/4)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.42.43.50:8123
DEBUG:urllib3.connectionpool:http://10.42.43.50:8123 "GET /api/states/sensor.dsmr_reader_dsmr_reader_current_l1 HTTP/11" 200 447
INFO:b2500-meter:Successfully fetched HomeAssistant powermeter value (filter 0.0.0.0/0): 4.0W
DEBUG:b2500-meter:Starting device: shellypro3em_old
DEBUG:b2500-meter:Shelly Pro 3EM Settings:
DEBUG:b2500-meter:Device ID: shellypro3em-ec4609c439c1
DEBUG:b2500-meter:Starting device: shellypro3em_new
DEBUG:b2500-meter:Shelly Pro 3EM Settings:
DEBUG:b2500-meter:Device ID: shellypro3em-ec4609c439c1
INFO:b2500-meter:Shelly emulator listening on UDP port 1010...
INFO:b2500-meter:Shelly emulator listening on UDP port 2220...
DEBUG:b2500-meter:Received UDP message: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
DEBUG:b2500-meter:From: 10.42.43.2:22222
DEBUG:b2500-meter:Parsed request: {
  "id": 0,
  "method": "EM.GetStatus",
  "params": {
    "id": 0
  }
}
DEBUG:urllib3.connectionpool:http://10.42.43.50:8123 "GET /api/states/sensor.dsmr_reader_dsmr_reader_current_l1 HTTP/11" 200 447
DEBUG:b2500-meter:Sending response: {"id":0,"src":"shellypro3em-ec4609c439c1","dst":"unknown","result":{"a_act_power":4.0,"b_act_power":0.001,"c_act_power":0.001,"total_act_power":4.001}}
DEBUG:b2500-meter:Received UDP message: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
DEBUG:b2500-meter:From: 10.42.43.2:22222
DEBUG:b2500-meter:Parsed request: {
  "id": 0,
  "method": "EM.GetStatus",
  "params": {
    "id": 0
  }
}

Scaro657, 2025-12-05:
Actually I just noticed you wrote "connected through wifi". So I suspect yours would be a different problem, since for me it worked even with the original (short) message when connected to wifi. I only had the problem when Venus was connected via LAN.

sln302, 2025-12-05:
Still something strange:

it's discharging to "max discharge power" instead of trying to go for NOM...

[image]